mavlink/mavlink #313: Mavgen does nothing when run

legonigel, 2015-01-13:
I ran the mavgen script from the command line in Linux (as described in the README.md):

```bash
$python -m pymavlink.generator.mavgen 
```

By default this should generate a python mavlink file, but I get no output.  I ensured that my PYTHONPATH contained the mavlink folder. Running the file gives no error and prints nothing. Maybe the docs just need some updating.

vooon, 2015-01-13:
Mavgen executable moved, please use:

    python -m pymavlink.tools.mavgen

legonigel, 2015-01-13:
This should be fixed in the docs. Also, old code should be removed or somehow cleaned up.
This worked for me:
```
$ python -m pymavlink.tools.mavgen
```

legonigel, 2015-01-13:
See pull request #314.

legonigel, 2015-01-18:
Fixed in commit 22ab243c8a1b24596ecc2714869333502331b869

closed 2015-01-18 by legonigel