mavlink/mavlink #337: Parameter protocol doesn't allow for useful errors

Susurrus, 2015-02-13:
For example I want to limit parameter setting to only certain modes of operation. There is no error emitting/handling built in to the parameter protocol, unlike with the mission protocol. I'd suggest the addition of a PARAM_ACK message with at least the following:
  * MAV_PARAM_ERROR - Generic error. Retrying may not resolve this.
  * MAV_PARAM_INVALID - The given parameter value is invalid.
  * MAV_PARAM_MISSING - The given parameter ID/name does not exist.
  * MAV_PARAM_DENIED - Setting parameters is currently disabled.

This message would be an appropriate response to any either PARAM_* message.

Susurrus, 2015-02-13:
And since this isn't part of QGC, which has a ridiculously long timeout on their parameter state machine, the user is left with no feedback as to what happened. I imagine more than just me want to implement this.

LorenzMeier, 2018-05-26:
Sounds like a QGC issue.

closed 2018-05-26 by LorenzMeier