mavlink/mavlink #339: mavutil.py autogeneration of Python libs doesn't guarantee up-to-date versions

Susurrus, 2015-02-18:
I've run into this problem several times when running mavloss.py where I end up seeing super-high loss rates (80%+). I've figured out this is from an auto-of-date Python implementation of my MAVLink messageset. mavutil generates these dialects, but it doesn't check if they're up-to-date. I'm wondering if a simple timestamp should be checked between the .xml files and the resulting .py files to see if they should be regenerated. Otherwise hashing the final dictionary of messages (after all the included .xml files are merged) and putting that in the resultant library might work well.

amilcarlucas, 2018-02-19:
AFAIK all files get re-generated. I do not see your point

Susurrus, 2018-02-19:
I no longer use this software, but I remember this biting me several times. I'll just close this as "won't fix, can't reproduce".

closed 2018-02-19 by Susurrus