mavlink/mavlink #347: Ambiguity between periods and frequencies in REQUEST_DATA_STREAM

phelpsw, 2015-03-04:
REQUEST_DATA_STREAM message rate  spec https://github.com/mavlink/mavlink/blob/b160f0e3c7174c115117eece93be691488faaf51/message_definitions/v1.0/common.xml#L1925 confuses frequencies and periods for specifying a rate.

The documentation for req_message_rate says: "The requested interval between two messages of this type" which would imply a period rather than a frequency.  Yet ardupilot for example interprets this as a frequency and not a period: https://github.com/diydrones/ardupilot/blob/2aae59437136fe610f8b899ac3fbd2111f3694b4/libraries/GCS_MAVLink/GCS_Common.cpp#L467

It would make sense to name this req_message_period perhaps to remove this ambiguity.  In my experience specifying a rate as a period in units of milli or microseconds works well for this sort of thing.

Susurrus, 2015-03-11:
Issue already reported: #4.

kd0aij, 2015-05-23:
#4's been open for more than 3 years???

pgiacalo, 2015-05-24:
Recent testing with both the Pixhawk/APM and the UDB/MatrixPilot autopilots shows that the field REQUEST_DATA_STREAM.req_message_rate is actually the requested message rate, in messages/second. So it's a frequency rather than the period. 

If the mavlink team addresses this item, then the documentation should probably say something like, "req_message_rate - the requested message rate, in messages/second". 

Providing the units is a simple way to avoid ambiguity and would be a good general practice for all mavlink fields that carry units.

LorenzMeier, 2015-08-10:
Will fix the docs, and this message will be deprecated anyway.

closed 2015-08-10 by LorenzMeier