mavlink/mavlink #350: ERROR While Generate C# Mavlink

xuhao1, 2015-03-16:
olivia_mav\c_sharp.generated.cs(983,39): error CS0031: 常量值“65536”无法转换为“ushort”
olivia_mav\c_sharp.generated.cs(988,36): error CS0031: 常量值“131072”无法转换为“ushort”
olivia_mav\c_sharp.generated.cs(993,37): error CS0031: 常量值“262144”无法转换为“ushort”
olivia_mav\c_sharp.generated.cs(998,35): error CS0031: 常量值“524288”无法转换为“ushort”
olivia_mav\c_sharp.generated.cs(1003,29): error CS0031: 常量值“1048576”无法转换为“ushort”
olivia_mav\c_sharp.generated.cs(1008,25): error CS0031: 常量值“2097152”无法转换为“ushort”
olivia_mav\c_sharp.generated.cs(1013,28): error CS0031: 常量值“4194304”无法转换为“ushort”
olivia_mav\c_sharp.generated.cs(1014,36): error CS0031: 常量值“4194305”无法转换为“ushort”

常量值“4194305”无法转换为“ushort” =
constant value “4194305 can't convert to “ushort”

SkiTiSu, 2015-04-24:
@xuhao1 Hi, 试试这样：
将mavgen_cs.py中第76行修改ushort为ulong，即
        outf.write("\tpublic enum %s : ulong \n\t{\n" % e.name)
编译通过

xuhao1, 2015-06-20:
@SkiTiSu TKS,我们现在改成用mavlink.net了

closed 2020-04-23 by hamishwillee