mavlink/mavlink #404: pymavlink doesn't work with parameters of type MAV_PARAM_TYPE_UINT32

arthurbenemann, 2015-06-20:
There is a conversion to float that happens when sending a param_set message with type `MAV_PARAM_TYPE_UINT32`. It likely happens to any messages with types different than REAL32

This is the line in question on the generated files:
```python
class MAVLink_param_set_message(MAVLink_message):
...
     def pack(self, mav):
                return MAVLink_message.pack(self, mav, 168, struct.pack('<fBB16sB', self.param_value, self.target_system, self.target_component, self.param_id, self.param_type))
```

arthurbenemann, 2015-06-20:
May I comment that this is a great number for an issue (#404)

amilcarlucas, 2018-02-21:
This is not a mavlink bug. It is a pymavlink bug and should therefor be reported upstream at: https://github.com/ArduPilot/pymavlink/issues and be closed here

closed 2018-02-21 by LorenzMeier