mavlink/mavlink #426: Clarify how to arm/disarm with the common messageset

mairsbw, 2015-08-18:
QGC uses the `SET_MODE` message, while there is also the `MAV_CMD_SET_MODE` command. Since there is overlap, I would think `SET_MODE` should be deprecated in favor of `COMMAND_LONG`. Though that message is a little longer, this is a rare message and it shouldn't be a problem.

LorenzMeier, 2015-08-18:
Could you send a PR in which it is deprecated in the comment?

mairsbw, 2015-08-18:
Sure. Should I file a ticket with QGC as well for them to update?

mairsbw, 2015-08-18:
Alright, but now I have a few follow up questions:
  1) Which command should it use, COMMAND_INT or COMMAND_LONG? Or are those messages interchangeable, it's not clear by their description.
  2) Should this be deprecated in favor of `MAV_CMD_DO_SET_MODE`? I believe it should, though this message should state that the component value will be ignored if used with a `COMMAND_*` message, since this is a system-wide attribute.
  3) Should `MAV_CMD_COMPONENT_ARM_DISARM` be deprecated? I didn't think the concept of "armed" or "disarmed" could apply to the component level, just the entire system. It's not used anywhere in the QGC codebase either.

mairsbw, 2015-08-18:
Note that `COMMAND_INT` isn't used anywhere in QGC as well.

LorenzMeier, 2015-08-18:
1) They are interchangeable and necessary because of the limitation of precision of float.
2) Yes, we should deprecate but not remove the message in favor of MAV_CMD
3) Not it should not, and we do commonly arm just gimbals and other connected devices.

mairsbw, 2015-12-01:
Why don't we just have COMMAND_LONG and pack an int into the 3 float fields that need higher precision and use a MAV_CMD_NAV_WAYPOINT_INT command to indicate this to the AP?

It's confusing what the purpose of both COMMAND_* messages are, given that a) their description doesn't really indicate which *should* be used and *when* and b) some fields are mission in COMMAND_LONG that exist in COMMAND_INT. Can you provide some more clarify on how these messages should be used? I'd like to add them into the common.xml descriptions for them.

mairsbw, 2015-12-01:
I realized that this still doesn't address the origin point of this issue: what is the proper way to arm/disarm the system? Is it through the `MAV_CMD_DO_SET_MODE` or is it through `MAV_CMD_COMPONENT_ARM_DISARM` with a component ID of 0 (or 255, I think one of these means all components)?

mairsbw, 2015-12-11:
@LorenzMeier @tridge Ping.

LorenzMeier, 2015-12-11:
Both are "proper". One is configuring the complete system, while the other one is just arming / disarming. If you don't know the current mode of the system using ARM_DISARM is beneficial as it avoids a race condition with setting the mode.

closed 2015-12-11 by LorenzMeier