mavlink/mavlink #431: LANDING_TARGET missing a field

djnugent, 2015-08-25:
I have realized that the LANDING_TARGET is missing an orientation field(yaw). @LorenzMeier @rmackay9 Is it too late to change it? I think I am still the only one who has uses this message so it shouldn't disrupt too much.

TSC21, 2015-08-26:
@djnugent I also pointed that out too. But, as @LorenzMeier said, it would be good to have a test point for the current implementation of the message and also a working framework on the firmware side (which I'm looking to get in here https://github.com/PX4/Firmware/pull/2446).
Anyhow, a quaternion field would be good in this case. Also I think this won't be a problem, as I think this is still in a dev state. But this is a @LorenzMeier and @rmackay9 call.

djnugent, 2015-08-26:
@TSC21 I am at the point where I am implementing the firmware side of this message and need orientation.

TSC21, 2015-08-26:
@djnugent follow up what I sent you above. Right now we need to add the usage. Mainly on the navigator side, so we can issue a landing command and, for example, if it detects valid `LANDING_TARGET` stream, it uses it's position to land. That also requires the usage of `local_position` data so it can land.
Continue the work on PX4/Firmware#2446 if you wish so by cloning the respective branch. Then you can push your modifications to it so discuss them and test their usability.

TSC21, 2015-08-26:
Also, if you are using ROS, this may become handy: https://github.com/mavlink/mavros/pull/311

rmackay9, 2015-08-26:
@djnugent, I think if you raise a PR with that field in it, it'll be accepted.  As you say, I don't think there are any consumers yet.  If @TSC21 wants it as well then that's another plus.

TSC21, 2015-08-26:
For me, it's a green light.

mhkabir, 2015-08-29:
Please have a look at my comments on the PR.

LorenzMeier, 2015-12-11:
Closing, to be discussed in the PR.

closed 2015-12-11 by LorenzMeier