mavlink/mavlink #437: tools directory is missing from package on PyPI

cqql, 2015-09-09:
Hello,
I just installed the `pymavlink` package from PyPI and tried to run `python -m pymavlink.tools.mavgen`, but that could not be found. Then I looked through the installed files, found the `generators` directory and tried the `mavgen` from there
```sh
$ python -m pymavlink.generator.mavgen
Traceback (most recent call last):
  File "/home/cqql/.pyenv/versions/3.5.0rc3/lib/python3.5/runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/cqql/.pyenv/versions/3.5.0rc3/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/cqql/.pyenv/versions/qopter/lib/python3.5/site-packages/pymavlink/generator/mavgen.py", line 167, in <module>
    raise DeprecationWarning("Executable was moved to pymavlink.tools.mavgen")
DeprecationWarning: Executable was moved to pymavlink.tools.mavgen
```

But there is no `tools` directory
```sh
$ tree ~/.pyenv/versions/qopter/lib/python3.5/site-packages/pymavlink
/home/cqql/.pyenv/versions/qopter/lib/python3.5/site-packages/pymavlink
├── a.py
├── DFReader.py
├── dialects
│   ├── __init__.py
│   ├── __pycache__
│   │   └── __init__.cpython-35.pyc
│   ├── v09
│   │   ├── ardupilotmega.py
│   │   ├── ardupilotmega.xml
│   │   ├── common.py
│   │   ├── common.xml
│   │   ├── __init__.py
│   │   ├── minimal.py
│   │   ├── minimal.xml
│   │   ├── pixhawk.py
│   │   ├── pixhawk.xml
│   │   ├── __pycache__
│   │   │   ├── ardupilotmega.cpython-35.pyc
│   │   │   ├── common.cpython-35.pyc
│   │   │   ├── __init__.cpython-35.pyc
│   │   │   ├── minimal.cpython-35.pyc
│   │   │   ├── pixhawk.cpython-35.pyc
│   │   │   ├── slugs.cpython-35.pyc
│   │   │   ├── test.cpython-35.pyc
│   │   │   └── ualberta.cpython-35.pyc
│   │   ├── slugs.py
│   │   ├── slugs.xml
│   │   ├── test.py
│   │   ├── test.xml
│   │   ├── ualberta.py
│   │   └── ualberta.xml
│   └── v10
│       ├── ardupilotmega.py
│       ├── ardupilotmega.xml
│       ├── ASLUAV.py
│       ├── ASLUAV.xml
│       ├── autoquad.py
│       ├── autoquad.xml
│       ├── common.py
│       ├── common.xml
│       ├── __init__.py
│       ├── matrixpilot.py
│       ├── matrixpilot.xml
│       ├── minimal.py
│       ├── minimal.xml
│       ├── pixhawk.py
│       ├── pixhawk.xml
│       ├── __pycache__
│       │   ├── ardupilotmega.cpython-35.pyc
│       │   ├── ASLUAV.cpython-35.pyc
│       │   ├── autoquad.cpython-35.pyc
│       │   ├── common.cpython-35.pyc
│       │   ├── __init__.cpython-35.pyc
│       │   ├── matrixpilot.cpython-35.pyc
│       │   ├── minimal.cpython-35.pyc
│       │   ├── pixhawk.cpython-35.pyc
│       │   ├── python_array_test.cpython-35.pyc
│       │   ├── sensesoar.cpython-35.pyc
│       │   ├── slugs.cpython-35.pyc
│       │   ├── test.cpython-35.pyc
│       │   └── ualberta.cpython-35.pyc
│       ├── python_array_test.py
│       ├── python_array_test.xml
│       ├── sensesoar.py
│       ├── sensesoar.xml
│       ├── slugs.py
│       ├── slugs.xml
│       ├── test.py
│       ├── test.xml
│       ├── ualberta.py
│       └── ualberta.xml
├── fgFDM.py
├── generator
│   ├── C
│   │   ├── include_v0.9
│   │   │   ├── checksum.h
│   │   │   ├── mavlink_helpers.h
│   │   │   ├── mavlink_types.h
│   │   │   └── protocol.h
│   │   └── include_v1.0
│   │       ├── checksum.h
│   │       ├── mavlink_conversions.h
│   │       ├── mavlink_helpers.h
│   │       ├── mavlink_types.h
│   │       └── protocol.h
│   ├── __init__.py
│   ├── java
│   │   └── lib
│   │       ├── Messages
│   │       │   ├── MAVLinkMessage.java
│   │       │   ├── MAVLinkPayload.java
│   │       │   └── MAVLinkStats.java
│   │       └── Parser.java
│   ├── lib
│   │   ├── genxmlif
│   │   │   ├── __init__.py
│   │   │   ├── __pycache__
│   │   │   │   ├── xmlifBase.cpython-35.pyc
│   │   │   │   └── xmlifODict.cpython-35.pyc
│   │   │   ├── xmlif4Dom.py
│   │   │   ├── xmlifApi.py
│   │   │   ├── xmlifBase.py
│   │   │   ├── xmlifDom.py
│   │   │   ├── xmlifElementTree.py
│   │   │   ├── xmlifMinidom.py
│   │   │   ├── xmlifODict.py
│   │   │   ├── xmliftest.py
│   │   │   └── xmlifUtils.py
│   │   ├── __init__.py
│   │   ├── minixsv
│   │   │   ├── datatypes.xsd
│   │   │   ├── __init__.py
│   │   │   ├── minixsvWrapper.py
│   │   │   ├── __pycache__
│   │   │   │   ├── __init__.cpython-35.pyc
│   │   │   │   └── xsvalXmlIf.cpython-35.pyc
│   │   │   ├── pyxsval.py
│   │   │   ├── XInclude.xsd
│   │   │   ├── XMLSchema-instance.xsd
│   │   │   ├── XMLSchema.xsd
│   │   │   ├── xml.xsd
│   │   │   ├── xsvalBase.py
│   │   │   ├── xsvalErrorHandler.py
│   │   │   ├── xsvalSchema.py
│   │   │   ├── xsvalSimpleTypes.py
│   │   │   ├── xsvalUtils.py
│   │   │   └── xsvalXmlIf.py
│   │   └── __pycache__
│   │       └── __init__.cpython-35.pyc
│   ├── mavcrc.py
│   ├── mavgen_c.py
│   ├── mavgen_cs.py
│   ├── mavgen_java.py
│   ├── mavgen_javascript.py
│   ├── mavgen_objc.py
│   ├── mavgen.py
│   ├── mavgen_python.py
│   ├── mavgen_wlua.py
│   ├── mavlink.py
│   ├── mavparse.py
│   ├── mavschema.xsd
│   ├── mavtemplate.py
│   ├── mavtestgen.py
│   └── __pycache__
│       ├── __init__.cpython-35.pyc
│       ├── mavcrc.cpython-35.pyc
│       ├── mavgen_c.cpython-35.pyc
│       ├── mavgen.cpython-35.pyc
│       ├── mavgen_cs.cpython-35.pyc
│       ├── mavgen_java.cpython-35.pyc
│       ├── mavgen_javascript.cpython-35.pyc
│       ├── mavgen_objc.cpython-35.pyc
│       ├── mavgen_python.cpython-35.pyc
│       ├── mavgen_wlua.cpython-35.pyc
│       ├── mavlink.cpython-35.pyc
│       ├── mavparse.cpython-35.pyc
│       ├── mavtemplate.cpython-35.pyc
│       └── mavtestgen.cpython-35.pyc
├── __init__.py
├── mavextra.py
├── mavlinkv09.py
├── mavlinkv10.py
├── mavnative
│   └── mavlink_defaults.h
├── mavparm.py
├── mavutil.py
├── mavwp.py
├── __pycache__
│   ├── a.cpython-35.pyc
│   ├── DFReader.cpython-35.pyc
│   ├── fgFDM.cpython-35.pyc
│   ├── __init__.cpython-35.pyc
│   ├── mavextra.cpython-35.pyc
│   ├── mavlinkv09.cpython-35.pyc
│   ├── mavlinkv10.cpython-35.pyc
│   ├── mavparm.cpython-35.pyc
│   ├── mavutil.cpython-35.pyc
│   ├── mavwp.cpython-35.pyc
│   ├── quaternion.cpython-35.pyc
│   ├── rotmat.cpython-35.pyc
│   └── setup.cpython-35.pyc
├── quaternion.py
├── rotmat.py
└── setup.py

22 directories, 162 files
```

vooon, 2015-09-09:
Tools installed as a scripts, so it is in `bin` directory (check your `~/.pyenv/versions/qopter/bin`).

But perhaps better to change this behaviour? Ping @LorenzMeier @tridge.

cqql, 2015-09-09:
Oh, I did not try that. I just tried to run the example commands from the readme. Just plain `mavgen.py messages.xml` works. It would be nice if the readme mentioned, that that is how it works if you install it from PyPI.

LorenzMeier, 2018-05-26:
Closing as stale.

closed 2018-05-26 by LorenzMeier