mavlink/mavlink #440: Request: Loiter Until Command

labtoast, 2015-10-03:
I propose a loiter until command that causes vehicle to wait until system time has elapsed commanded time. This would be used for synchronizing a patrol path or synchronizing movements between multiple vehicles in an open loop manner. 

```
<entry value="19" name="MAV_CMD_NAV_LOITER_UNTIL">
                    <description>Loiter around this MISSION until X time</description>
                    <param index="1">Seconds (decimal)</param>
                    <param index="2">0 - time since boot; 1 - absolute time</param>
                    <param index="3">Radius around MISSION, in meters. If positive loiter clockwise, else counter-clockwise</param>
                    <param index="4">Desired yaw angle.</param>
                    <param index="5">Latitude</param>
                    <param index="6">Longitude</param>
                    <param index="7">Altitude</param>
               </entry>
```

LorenzMeier, 2015-12-11:
This is already present.

closed 2015-12-11 by LorenzMeier

labtoast, 2015-12-11:
I don't see it anywhere in `common.xml`

The closest command is LOITER_TIME  which lets you *delay* by a period of time. The purpose of a LOITER_UNTIL is to synchronize a path with a clock. I would like to perform a movement or action once per minute. The action can take a variable amount of time so a loiter until would synchronize this.

```
<entry value="19" name="MAV_CMD_NAV_LOITER_TIME">
                    <description>Loiter around this MISSION for X seconds</description>
                    <param index="1">Seconds (decimal)</param>
                    <param index="2">Empty</param>
                    <param index="3">Radius around MISSION, in meters. If positive loiter clockwise, else counter-clockwise</param>
                    <param index="4">Desired yaw angle.</param>
                    <param index="5">Latitude</param>
                    <param index="6">Longitude</param>
                    <param index="7">Altitude</param>
               </entry>
```

magicrub, 2015-12-11:
does time<=0 mean forever? If so then your "command" is to just skip to the next waypoint

labtoast, 2015-12-11:
`                    <param index="2">0 - time since boot; 1 - absolute time</param>`

I imagined that the inputs would be positive only. It would take UNIX time or time since boot of the flight controller mcu. If the input time was less than current time then the command would skip like you say.  Otherwise, a loiter would occur until the specified time.

labtoast, 2015-12-16:
Is this feature already present? Can we reopen this issue if not?