mavlink/mavlink #442: Format XML files using xmllint (libxml2 package)

tcr3dr, 2015-10-09:
I started work on upstreaming the MAVLink changes that Solo introduces to ardupilotmega.xml. The biggest issue in several months' of development work, apparently, is whitespace issues that have accumulated.

At least ardupilotmega.xml both upstream and downstream have their own whitespace issues (inconsistent mixed tabs and spaces). Though it makes `git blame` more complicated (requiring the `-w` flag and rendering Github's git blame view less useful), I propose we run all the XML files through `xmllint --format`—with no additional command line flags—and require this for additional changes, to simplify the process in the future.

`xmllint` is included with OS X.
`apt-get install libxml2-utils` for Ubuntu distributions.

/cc @tridge

mrpollo, 2015-10-09:
My biggest issue would be how to enforce this, if we can just ask devs to do it and this is acceptable then I would vote in favor, what do you guys think @LorenzMeier @tridge

vooon, 2015-10-09:
I think Travis can check formatting. Similar thing done for [rosdistro](https://github.com/ros/rosdistro).

tcr3dr, 2015-10-13:
Enforcement via a simple Python script (good): https://github.com/ros/rosdistro/blob/master/test/rosdistro_formatting_test.py

amilcarlucas, 2018-02-19:
This has already been implemented. And the ardupilot fork is now closer to the mavlin/mavlink master as well.

This can be closed now

hamishwillee, 2018-02-19:
Agreed - yes, is run on travis via script  https://github.com/mavlink/mavlink/blob/master/scripts/format_xml.sh. Thanks @amilcarlucas

closed 2018-02-19 by hamishwillee