mavlink/mavlink #443: Linker error on Mac OS with Anaconda

jacobsenmd, 2015-10-10:
I have been using Python Anaconda (2.7, 64-bit) so I can have a consistent Python environment across Windows/Mac OS/Linux with easy package management, but I get a linker error when running pymavlink's 'setup.py install' or running 'pip install pymavlink'. When I disable Anaconda and revert back to the Mac OS Python 2.7 installation it builds/installs fine. It also builds/installs fine using Anaconda on Windows 10 and Ubuntu 14. I don't know much at all about gcc/clang, but can only guess that Anaconda on Mac OS is somehow using a non-different version of gcc than everything else, and that this version handles warnings differently. Output below.

I realize this is a pretty specific issue, but given Anaconda's popularity, it would be great if the pymavlink installer would work with it. Thanks! 

building 'mavnative' extension
gcc -fno-strict-aliasing -I//anaconda/include -arch x86_64 -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -Igenerator/C/include_v1.0 -Imavnative -I//anaconda/include/python2.7 -c mavnative/mavnative.c -o build/temp.macosx-10.5-x86_64-2.7/mavnative/mavnative.o
mavnative/mavnative.c:155:13: warning: enumeration value
      'MAVLINK_PARSE_STATE_GOT_BAD_CRC1' not handled in switch [-Wswitch]
    switch (status->parse_state)
            ^
1 warning generated.
gcc -bundle -undefined dynamic_lookup -L//anaconda/lib -arch x86_64 -arch x86_64 build/temp.macosx-10.5-x86_64-2.7/mavnative/mavnative.o -L//anaconda/lib -o build/lib.macosx-10.5-x86_64-2.7/mavnative.so
ld: library not found for -lgcc_s.10.5
clang: error: linker command failed with exit code 1 (use -v to see invocation)
error: command 'gcc' failed with exit status 1

jacobsenmd, 2015-10-11:
Relevant link (I think) here: http://clang-developers.42468.n3.nabble.com/Question-on-Wswitch-enum-td4025927.html

amueller, 2016-03-24:
weird (kind of) fix here: http://flothesof.github.io/sms-tools-MacOSX-Anaconda.html

amilcarlucas, 2018-02-21:
this is a pymavlink bug. Please report it upstream at https://github.com/ArduPilot/pymavlink/issues

closed 2018-02-21 by LorenzMeier