mavlink/mavlink #450: Region of interest MAV_CMD_NAV_ROI

jlesk, 2015-10-30:
hello,
I don't understand how MAV_CMD_NAV_ROI should work.
When this command is sent to a drone should it start to watch the ROI since the next waypoint or the previous one?
If we send this command twice in a row, should we look only the last received ROI?
Or during the flightplan, the command MAV_CMD_DO_SET_ROI should be sent to the drone to select one of the  previously transmitted ROI?
Maybe it is up to the drone maker to decide?

amilcarlucas, 2018-02-19:
Yes, it is for the drone programmer/maker to decide.

That is a support question. But this forum is for bug reports...

IMHO this should be closed.

hamishwillee, 2018-02-20:
Thanks @amilcarlucas - yes, closing.

closed 2018-02-20 by hamishwillee