mavlink/mavlink #455: Issue in the mobcap mavlink message, NO target_system !!

mzahana, 2015-11-07:
Hi,

Why there is no a field for target_system in the ATT_POS_MOCAP mavlink message?
Suppose I am sending 3 mocap messages to 3 different quads, at the same time, which have different sys_id, how can I do this with the current message? 
When I tried to send mocap info to different quads, they all received AND used the same mocap message to estimate position and attitude, which does not make sense.
I believe there should be a filed for target_system, in order to differentiate between different Mavs.

Also there will be need to do a modification in the mavlink_receiver.cpp in order to handle this, too.
What do you think?

```html
<message id="138" name="ATT_POS_MOCAP">
          <description>Motion capture attitude and position</description>
          <field type="uint64_t" name="time_usec">Timestamp (micros since boot or Unix epoch)</field>
          <!--the new field-->
          <field type="uint8_t" name="target_system">Target system ID: 1~255</field>*
          <field type="float[4]" name="q">Attitude quaternion (w, x, y, z order, zero-rotation is 1, 0, 0, 0)</field>
          <field type="float" name="x">X position in meters (NED)</field>
          <field type="float" name="y">Y position in meters (NED)</field>
          <field type="float" name="z">Z position in meters (NED)</field>
        </message>
```

amilcarlucas, 2018-03-08:
Please add a PR for this

LorenzMeier, 2018-03-08:
Closing this as the PR would take care of it.

closed 2018-03-08 by LorenzMeier