mavlink/mavlink #458: C Include guards exclude other mavlik message specification

ptsneves, 2015-11-10:
I have a situation where I use Mavlink messages to serialize and transmit data across 2 MAV components. This communication uses it's own xml file and thus it has it's own generated headers. 

It happens that these MAV use MAVLink for normal communication with the common.xml message set. The message names overlap (for instance have a heartbeat message with different parameters), so they cannot be merged into the same XML. Even if it was possible it would be very strange because they relate to different functionality.

So I decided my library, which uses MAVLink, would also include it's version of the generated headers for my messages. The problem is that due to no namespacing in the include guards' name my headers do not get included. 

I am working with adding the basename to a the prefix of the include guards and the functions but it gets very messy fast.

What is the community opinion on such a patch?

Susurrus, 2015-11-11:
I'd welcome a solution to this, but function name overloading isn't possible in C with just having 1 global namespace. You might be able to do something by generating a lookup table of function calls that's indexed by the messageset ID effectively. Do you have any ideas on how you'd implement this?

ptsneves, 2015-11-16:
My solution that I tried but was flawed was to add a _<xml_basename>_ to the filenames, message names and include guards. The problem is that then I also have to change the supposedly static protocol includes. It went nowhere. 

I am not sure I understand so well the how it would help the lookup table. The problem is the messages are directly coupled to the internals of the Mavlink protocol. Some work in decoupling this would probably be the first step.

I propose that the library users include a header with the name of the xml, and not a generic mavlink.h which includes the message set. This because mavlink.h will have include guards which are transversal to all generated  sets naturally creating an unnecessary namespacing conflict.

amilcarlucas, 2018-02-19:
Just include ardupilotmega.h and do not include common.h

This is IMHO not a bug and can be closed

hamishwillee, 2018-02-20:
@amilcarlucas Agreed. In any case it is pretty old. Posters, feel free to reopen if you think this is an issue still after reading https://github.com/mavlink/mavlink/issues/458#issuecomment-366713969

closed 2018-02-20 by hamishwillee

ptsneves, 2018-02-20:
@amilcarlucas I do not understand how your solution is valid for this problem. I said I am only using the common message set, and not anything related to ardupilotmega. Moreover, including ardupilotmega.h will result in the exact same problem. Can you clarify?

Also, is Mavlink now only for use in Ardupilot?

amilcarlucas, 2018-02-20:
Then instead of using `ardupilotmega.h` use `minimal.h`
And no, mavlink is not only for ardupilot, it is for a lot of different autopilot softwares, and can be seen on the names of the generated `.h` files. I just pointed at `ardupilotmega.h` as an example. But in your case `minimal.h` is the way to go.