mavlink/mavlink #473: Clarify *ATTITUDE_TARGET and similar messages

mairsbw, 2015-12-03:
What's the difference between `SET_ATTITUDE_TARGET` and `ATTITUDE_TARGET`? It sounds like `SET_ATTITUDE_TARGET` is a vehicle command and `ATTITUDE_TARGET` is the system response or status update. The descriptions are not different between them, however. So what the use case here? The system receives `SET_ATTITUDE_TARGET` messages and after each one responds with the `ATTITUDE_TARGET` message? Or is it supposed to be a periodic message when the vehicle is in the SET_ATTITUDE mode?

I have this same question about the `POSITION_TARGET_GLOBAL_INT` messages and the `POSITION_TARGET_LOCAL_NED` messages as well.

As a final clarification question about these messages, are they intended to be alternatives to manual control? So they should only be acted upon when the system is in the `MAV_MODE_FLAG_STABILIZE_ENABLED` or `MAV_MODE_FLAG_MANUAL_INPUT_ENABLED` modes?

mairsbw, 2015-12-11:
@LorenzMeier Any advice here? I'll offer up a PR with more clarify about these messages if someone can provide some more details.

LorenzMeier, 2015-12-11:
SET_* does command it, the other messages report it. The reporting happens independent of the command, as the system ALWAYS has an ATTITUDE_TARGET. These messages should only be acted upon if the system has the OFFBOARD_CONTROL (which is not a standard MAVLink thing yet) state active where its not generating references locally but accepts any of these commands.

mairsbw, 2015-12-11:
Can we pull `OFFBOARD_CONTROL` into standard MAVLink then? Sounds like this group of messages are incomplete without it. Is it just another MODE flag?

bwmairs, 2015-12-14:
Additionally there seems to be a large overlap in the functionality of `SET_POSITION_TARGET_LOCAL_NED` and `SET_POSITION_TARGET_GLOBAL_INT` as both need a 3-tuple to represent position and the `coordinate_frame` field can specify what field these are in. Why not just use a `SET_POSITION_TARGET` message and then let the frame of reference be specified by that field?

And since we're on the subject of external autopilot commands, what is `MANUAL_SETPOINT` for? When would this be used over the `MANUAL_CONTROL` or the `SET_ATTITUDE_TARGET` messages?

Susurrus, 2016-01-06:
@LorenzMeier Any updates here? Still trying to understand how the `MANUAL_SETPOINT`, `MANUAL_CONTROL`, and `SET_ATTITUDE_TARGET` messages all differ in usage.

amilcarlucas, 2018-02-21:
What is missing is a xml attribute for every message that states the intended direction:
* MAV_TO_GCS
* GCS_TO_MAV
* BIDIR

hamishwillee, 2020-04-23:
Closing. Original issues addressed

closed 2020-04-23 by hamishwillee