mavlink/mavlink #474: Java: buffer overflow while decoding message

WickedShell, 2015-12-06:
Basically it's trying to add a byte to the payload buffer when it runs out of space. Unfortunately this seems to have been caused by an error over the wire and I haven't been able to recreate it with a debugger. I'm not sure how we managed to exceed the buffer length anywhere.

Exception in thread "COM8" java.nio.BufferOverflowException
        at java.nio.Buffer.nextPutIndex(Buffer.java:521)
        at java.nio.HeapByteBuffer.put(HeapByteBuffer.java:169)
        at com.MAVLink.Messages.MAVLinkPayload.add(MAVLinkPayload.java:25)
        at com.MAVLink.Parser.mavlink_parse_char(Parser.java:79)
        at swiftgcs.mavparser$parseBytes.invoke(mavparser.clj:124)
        at serial.core$listen$reify__10359.serialEvent(core.clj:117)
        at purejavacomm.PureJavaSerialPort.sendDataEvents(PureJavaSerialPort.java:96)
        at purejavacomm.PureJavaSerialPort.access$2500(PureJavaSerialPort.java:47)
        at purejavacomm.PureJavaSerialPort$3.run(PureJavaSerialPort.java:1244)
        at java.lang.Thread.run(Thread.java:745)

magicrub, 2015-12-06:
Maybe the length value changed over the air? Sounds like low hanging fruit
which should have been checked for though.

WickedShell, 2015-12-06:
USB, and it happened almost immediately, but I agree there is no sanity
checking anywhere on the length and that would easily solve the problem.

WickedShell, 2016-01-02:
Alright managed to recreate this and attatch a debugger, and get a heap dump. It appears that the error is from https://github.com/mavlink/mavlink/blob/master/pymavlink/generator/java/lib/Parser.java#L97 basically if we didnt get they byte we expected for the CRC we check if its the start of a new packet. And then reset the state. This bypasses the normal resetting of the state that happens here https://github.com/mavlink/mavlink/blob/master/pymavlink/generator/java/lib/Parser.java#L46 which clears out the payload. This should be easy to correct, I'll get a PR in within the next couple of days.

I also noticed that the payload size is set to 512 bytes, which is not a valid payload length (the max valid payload length is 255 so we are wasting a fair amount of space). I'll probably get a PR in for this at the same time.

WickedShell, 2016-01-02:
Oh and I do have a heap dump saved for inspecting this if anyone really wants to look at it.

WickedShell, 2016-03-22:
The fix went in so closing the issue.

closed 2016-03-22 by WickedShell