mavlink/mavlink #482: C# generator problem

harewei, 2015-12-16:
I've tried to use mavgenerate.py GUI to generate C# library
XML: mavlink-master/message_definitions/v1.0/ardupilotmega.xml
however, I always get the error not finding source file for CS\common\ByteArrayUtil.cs, FrameworkBitConverter.cs and MavLink.cs, ever though they clearly exist

I tried generating for C and Java, they all work fine.

amilcarlucas, 2018-02-19:
Retest on master

hamishwillee, 2018-02-20:
@amilcarlucas This is still valid. I just ran it on Windows:
```
This compiler is provided as part of the Microsoft (R) .NET Framework, but only supports language versions up to C# 5, which is no longer the latest version. For compilers that support newer versions of the C# programming language, see http://go.microsoft.com/fwlink/?LinkID=533240

error CS2001: Source file 'CS\common\ByteArrayUtil.cs' could not be found
error CS2001: Source file 'CS\common\FrameworkBitConverter.cs' could not be found
error CS2001: Source file 'CS\common\Mavlink.cs' could not be found
Error
Error: Compilation failed. (1)
```
It may be that I'm missing some dependency.

hamishwillee, 2018-02-20:
Duplicated https://github.com/mavlink/mavlink/issues/813

amilcarlucas, 2018-02-20:
@hamishwillee This is a pymavlink issue that should be reported upstream to https://github.com/ardupilot/pymavlink/issues and closed here

hamishwillee, 2018-02-20:
Thanks @amilcarlucas - I've cross posted to https://github.com/ArduPilot/pymavlink/issues/151 and will now close.

closed 2018-02-20 by hamishwillee