mavlink/mavlink #484: Need support for multiple dialects within a single compiled codebase

DonLakeFlyer, 2015-12-19:
Since mavlink helpers compile in the CRCs and message lengths from the dialect being used you cannot create a single compiled codebase which uses more than one mavlink dialect. 

Possible solution: Provide callback mechanism which provides correct CRC and friends information based on vehicle id.

Susurrus, 2015-12-19:
Are you specifically referring to the C implementation? There's a PR for a C++ implementation, and I imagine that language would make it easier to implement this functionality.

DonLakeFlyer, 2015-12-19:
It's going to be a problem with all implementations, hence all need provide the support. I've only looked at the C code. In C++ you could override a base class virtual or something instead of a callback.

hamishwillee, 2020-04-23:
This would be cool, as would some way to easily support multiple versions. But closing because this is clearly not important enough to drive someone doing actual development.

closed 2020-04-23 by hamishwillee