mavlink/mavlink #486: Mavlink API in c/c++ can be developed in linux platform

CrazyPopLin, 2015-12-30:
Hi, I am trying to use a linux platform (arm board) to communicate APM stack Pixhawk. 
I want to use c/c++ API to send and receive message after include the mavlink header file. Is that any examples about that ? I know the python API like dronekit-python.
I am going through the qgroundcontrol (may be deveopled for px4 stack but still the example for communicating) linux source code which is c/c++. But it has too much stuff. 
Any suggestions will be appreciate and thanks for your help.

magicrub, 2015-12-30:
Hi @CrazyPopLin , you've posted a support question in the developer TODO list. Please visit http://www.mavlink.org/mavlink/start for info. You can also ask questions on the gitter chat channel: https://gitter.im/mavlink/qgroundcontrol

CrazyPopLin, 2016-01-04:
found [c_uart_interface_example] (https://github.com/mavlink/c_uart_interface_example)

closed 2016-01-04 by CrazyPopLin