mavlink/mavlink #490: Couldn't find MAV_NAV_MODE enum or message definitions.

NicksonYap, 2016-01-03:
Hi,
I'm developing a TX RX radio based on NRF24L01+ where telemetry data will be sent to OpenTX remote (Flysky TH-9X)
So far am able to get telemetry data displayed on screen using MAVLink (ardupilotmega)
I ran into `mavlink_msg_hil_controls_pack` and checked the message definitios for "nav_mode"

```C++
/**
 * @brief Pack a hil_controls message
 * @param system_id ID of this system
 * @param component_id ID of this component (e.g. 200 for IMU)
 * @param msg The MAVLink message to compress the data into
 *
 * @param time_usec Timestamp (microseconds since UNIX epoch or microseconds since system boot)
 * @param roll_ailerons Control output -1 .. 1
 * @param pitch_elevator Control output -1 .. 1
 * @param yaw_rudder Control output -1 .. 1
 * @param throttle Throttle 0 .. 1
 * @param aux1 Aux 1, -1 .. 1
 * @param aux2 Aux 2, -1 .. 1
 * @param aux3 Aux 3, -1 .. 1
 * @param aux4 Aux 4, -1 .. 1
 * @param mode System mode (MAV_MODE)
 * @param nav_mode Navigation mode (MAV_NAV_MODE)
 * @return length of the message in bytes (excluding serial stream start sign)
 */
static inline uint16_t mavlink_msg_hil_controls_pack(uint8_t system_id, uint8_t component_id, mavlink_message_t* msg, uint64_t time_usec, float roll_ailerons, float pitch_elevator, float yaw_rudder, float throttle, float aux1, float aux2, float aux3, float aux4, uint8_t mode, uint8_t nav_mode)
{...}
```

I used Notepadqq (a clone of Notepad++ for Linux) to search through the source from repos: mavlink/c_liibrary and mavlink/mavlink
And even manually look from "mavlink_types.h"

And found no description about MAV_NAV_MODE.

Weirdly, openTX declares a variable called "nav_mode" but then gets it's data from "mav_mode" instead of "mav_nav_mode".

I assume "mav_nav_mode" is not used or controller-specific?
So I should just continue with "mav_mode" instead?

Thanks!

amilcarlucas, 2018-02-19:
This is for bug reports, not for support questions.

IMHO this can be closed now.

hamishwillee, 2018-02-19:
Agreed. 

@NicksonYap FYI, the place for support is linked from the guide here: https://mavlink.io/en/#support

closed 2018-02-19 by hamishwillee

hamishwillee, 2024-12-18:
Actually, I was wrong. Very sorry. This is generated code, so the error is in mavlink managed definitions.

The answer to anyone coming here is "I don't know". Looking into it.