mavlink/mavlink #496: Shortcomings of the current GPS_RAW_INT and GPS2_RAW messages

WickedShell, 2016-01-06:
One of the problems we have with APM is that while we allow and use multiple GPS units there is no method for us to report over MAVLink which one is the primary GPS and being used to control the aircraft. If we could add a bool for isPrimary to both GPS_RAW_INT ( 24 ) and GPS2_RAW ( 124 ) that would be extremely helpful.

I think it's also worth mentioning that the differences in fields between the two, (GPS2 has DGPS reports). This actually adds complexity as if you wanted to use any of the DGPS reporting you have to ensure that the GPS is communicated as GPS2. This also imposes an artificial limit on the number of GPS instances that can be reported at once, as there are only 2 messages available to pass GPS data down to the GCS, and they are formatted differently and must be handled differently in the generation on the vehicle and the parser on the GCS or other target MAVLink device. I think we could clean both of these problems up quite easily by moving to a single GPS_RAW message that has a 8 bit field for tracking which GPS instance it is. This simplifies the parsing and the generation to always use the same code path quite nicely.

LorenzMeier, 2016-03-10:
@WickedShell Could you propose a new message format for this?

WickedShell, 2016-03-14:
I'd really like to move to just one GPS_RAW_INT message with a field used to indicate which instance it is. (Basically this allows the simplification and reduction of code as more data can follow a common path)

One of the initial things I really wanted was a field indicating if the GPS was being used as the primary instance, but one of the things thats been realized from talking with Paul, has been that it actually makes more sense to always give one GPS to a given EKF instance. So rather then just a primary field I think it makes sense to offer a bitmask of which EKF/AHRS instances are being feed with the GPS data. (And until the one GPS per instance is implemented if a GPS was considered primary then it could mark itself as providing all the instances).

I think the DGPS fields can be dropped as I can't seem to find anyone using them right now. (APM nil's the fields out, and I didn't find any references in PX4 for them). I can see that we could want to report this information, but I don't know what I've done enough with RTK/DGPS setups to know what should be reported here.

Something I've been thinking about recently, but I don't think it really belongs in the GPS message (maybe as part of the system health bit for GPS instead) is that the u-blox driver on APM now fully checks that every setting it sets has been set on the GPS, and reports it as an arming check (as well as polling periodically to keep checking). It would be worth knowing if the GPS isn't done verifying itself (or ever fails a periodic poll), but I think this can probably be handled in the SYS_STATUS health message instead. I've also been thinking about playing with some more complicated checks to report when the GPS has poor health/should be considered to be bad, but again just flipping the SYS_STATUS bit would probably cover this (and the information to report anything more detailed would probably be exceeding the detail to easily fit it in or justify putting it in the routinely sent message).

My proposed message would be (mostly just adding on top of the old GPS_RAW_INT, bold for my proposed additions)
               **<field type="uint8_t" name="instance">Which GPS instance is being reported</field>**
               <field type="uint64_t" name="time_usec">Timestamp (microseconds since UNIX epoch or microseconds since system boot)</field>
               <field type="uint8_t" name="fix_type">0-1: no fix, 2: 2D fix, 3: 3D fix, 4: DGPS, 5: RTK. Some applications will not use the value of this field unless it is at least two, so always correctly fill in the fix.</field>
               <field type="int32_t" name="lat">Latitude (WGS84), in degrees * 1E7</field>
               <field type="int32_t" name="lon">Longitude (WGS84), in degrees * 1E7</field>
               <field type="int32_t" name="alt">Altitude (AMSL, NOT WGS84), in meters * 1000 (positive for up). Note that virtually all GPS modules provide the AMSL altitude in addition to the WGS84 altitude.</field>
               <field type="uint16_t" name="eph">GPS HDOP horizontal dilution of position (unitless). If unknown, set to: UINT16_MAX</field>
               <field type="uint16_t" name="epv">GPS VDOP vertical dilution of position (unitless). If unknown, set to: UINT16_MAX</field>
               <field type="uint16_t" name="vel">GPS ground speed (m/s * 100). If unknown, set to: UINT16_MAX</field>
               <field type="uint16_t" name="cog">Course over ground (NOT heading, but direction of movement) in degrees * 100, 0.0..359.99 degrees. If unknown, set to: UINT16_MAX</field>
               <field type="uint8_t" name="satellites_visible">Number of satellites visible. If unknown, set to 255</field>
               **<field type="uint8_t" name="source">Which AHRS systems the GPS is acting as a source for</field>**

I'm definitely interested in what other people think though.

LorenzMeier, 2016-04-19:
@WickedShell We can't extend a message without breaking compatibility in MAVLink 1. So we would need to move to MAVLink 2 for that.

WickedShell, 2016-04-22:
@LorenzMeier Yeah, that was expected when I opened the initial issue. The issue was opened to track it, particularly with an aim of starting a discussion for mavlink2.

hamishwillee, 2019-02-17:
@WickedShell 

So to summarise, the message you are proposing is:
- two new extension fields to [GPS_RAW_INT](https://mavlink.io/en/messages/common.html#GPS_RAW_INT) - a GPS instance and an AHRS source.
- deprecating [GPS2_RAW](https://mavlink.io/en/messages/common.html#GPS2_RAW)?

1. I presume you are still interested in this? 
2. Anything else you would change given this was posted in 2016.
3. So field with new instance field shown below - there have already been some extensions added. is that OK?
```xml
    <message id="24" name="GPS_RAW_INT">
      <description>The global position, as returned by the Global Positioning System (GPS). This is
                NOT the global position estimate of the system, but rather a RAW sensor value. See message GLOBAL_POSITION for the global position estimate.</description>
      <field type="uint64_t" name="time_usec" units="us">Timestamp (UNIX Epoch time or time since system boot). The receiving end can infer timestamp format (since 1.1.1970 or since system boot) by checking for the magnitude the number.</field>
      <field type="uint8_t" name="fix_type" enum="GPS_FIX_TYPE">GPS fix type.</field>
      <field type="int32_t" name="lat" units="degE7">Latitude (WGS84, EGM96 ellipsoid)</field>
      <field type="int32_t" name="lon" units="degE7">Longitude (WGS84, EGM96 ellipsoid)</field>
      <field type="int32_t" name="alt" units="mm">Altitude (MSL). Positive for up. Note that virtually all GPS modules provide the MSL altitude in addition to the WGS84 altitude.</field>
      <field type="uint16_t" name="eph">GPS HDOP horizontal dilution of position (unitless). If unknown, set to: UINT16_MAX</field>
      <field type="uint16_t" name="epv">GPS VDOP vertical dilution of position (unitless). If unknown, set to: UINT16_MAX</field>
      <field type="uint16_t" name="vel" units="cm/s">GPS ground speed. If unknown, set to: UINT16_MAX</field>
      <field type="uint16_t" name="cog" units="cdeg">Course over ground (NOT heading, but direction of movement) in degrees * 100, 0.0..359.99 degrees. If unknown, set to: UINT16_MAX</field>
      <field type="uint8_t" name="satellites_visible">Number of satellites visible. If unknown, set to 255</field>
      <extensions/>
      <field type="int32_t" name="alt_ellipsoid" units="mm">Altitude (above WGS84, EGM96 ellipsoid). Positive for up.</field>
      <field type="uint32_t" name="h_acc" units="mm">Position uncertainty. Positive for up.</field>
      <field type="uint32_t" name="v_acc" units="mm">Altitude uncertainty. Positive for up.</field>
      <field type="uint32_t" name="vel_acc" units="mm">Speed uncertainty. Positive for up.</field>
      <field type="uint32_t" name="hdg_acc" units="degE5">Heading / track uncertainty</field>
```
```xml
      <field type="uint8_t" name="gps_instance">GPS instance being reported (1, 2, etc.)</field>
```
```xml
    </message>
```

4. What would the "Which AHRS systems the GPS is acting as a source for" look like? Can you propose the field and enum values? Or is this a bitmask and the GPS might act for multiple sources?

WickedShell, 2019-02-17:
> 1. I presume you are still interested in this?

Somewhat yes. I definitely prefer to see instance based messages rather then have the 2 3 4 variants, as well as the fact that there are differences in reportable data. (IE compare `GPS_RAW_INT` to `GPS2_RAW`)

> 2. Anything else you would change given this was posted in 2016.

If we made a new message `satellites_visible` should actually be satellites in use rather then visible. I use GPS's which track the two independently and in general you want to know how many are being used, rather then tracked. We can't retroactively fix this. Autopilots can make the decision to report in use here instead though if they prefer)

> 3.  So field with new instance field shown below - there have already been some extensions added. is that OK?

The instance needs to be counted from 0 so that extensions/MAVLink1 report correctly.

> 4. What would the "Which AHRS systems the GPS is acting as a source for" look like? Can you propose the field and enum values? Or is this a bitmask and the GPS might act for multiple sources?

This doesn't really belong in the GPS message actually. GPS reporting should stick to just the GPS attributes. The desire for the reporting is that we frequently see the issue with sensor switching that we don't know if the GPS is actually being used to fly the vehicle or is the backup. That reporting really belongs to the sensor fusion/AHRS though, rather then on the GPS message.

hamishwillee, 2019-02-18:
> The instance needs to be counted from 0 so that extensions/MAVLink1 report correctly.

I think you're saying that if the message is sent by a MAVLink 1 system the field will be received as 0 by the receiving mavlink 2 system? So we need to use zero.

> If we made a new message `satellites_visible` should actually be satellites in use rather then visible. I use GPS's which track the two independently and in general you want to know how many are [...]

Can/should we add extension `satellites_in_use` too. 

I didn't realise we were proposing a new message, presumably that combines anything from both the others?

Can you come to the meeting this week to discuss?

WickedShell, 2019-02-18:
> I think you're saying that if the message is sent by a MAVLink 1 system the field will be received as 0 by the receiving mavlink 2 system? So we need to use zero.

Yeah. The good news is from a programmer perspective arrays start at 0, and the message is still correct.

> I didn't realise we were proposing a new message, presumably that combines anything from both the others?

The bad news with an extension, and this should maybe be considered a hard block to the extension: Any system that doesn't understand/get sent the extensions will be very confused when it receives alternating GPS_RAW_INT for 2 different GPS's on the vehicles and their numbers/fixes look very different. This can easily result in a reported GPS mode flickering on the GCS.

> Can/should we add extension `satellites_in_use` too.

Probably not. Not enough systems report it/most users won't appreciate the difference. I'm not sure that growing the message with something that will be underused is all that valuable.

> Can you come to the meeting this week to discuss?

I'm going to try, but I have a meeting that morning ~16 hours before the MAVLink call, so I may be falling asleep/off by that point unfortunately. Will try and keep you posted.