mavlink/mavlink #498: pymavlink bug: serial.setBaudrate() is out of date

ChaoyuChen, 2016-01-07:
### Symptoms
I used mavproxy to trace mavlink with the command `mavproxy.py --master=/dev/ttyUSB0 --baudrate 57600 --aircraft MyCopter`. Then I got the error messages as below:
> Failed to connect to /dev/ttyUSB0 : 'Serial' object has no attribute 'setBaudrate'

### Analysis
In new version of python, the member funciton `setBaudrate()` had been removed from class `Serial`. We could directly assign `Serial.baudrate`. 

### Current Workaround
Modify the code piece in `pymavlink/mavutil.py`
>    def set_baudrate(self, baudrate):
        '''set baudrate'''
        self.port.setBaudrate(baudrate)

with

>    def set_baudrate(self, baudrate):
        '''set baudrate'''
        self.port.baudrate = baudrate

Then mavproxy worked correctly.

Susurrus, 2016-01-07:
You likely want to put this in a try..except so that both versions of pyserial can be supported.

tridge, 2016-01-28:
thanks for reporting this! I have pushed a new version that uses try/except to fix it

closed 2016-01-28 by tridge

M4gicT0, 2018-02-15:
Thank you soooo much for this fix !! It helped me fix the same bug in a completely different thing (tinyOS and msp430 for the TelosB).