mavlink/mavlink #509: Proper Follow-Me API for common.xml

LorenzMeier, 2016-02-15:
Here is an initial proposal for a follow-me message:

@liamstask @rmackay9 @jschall This is to support the addition of Follow-Me in QGroundControl (https://github.com/mavlink/qgroundcontrol/pull/2817) and I wanted to coordinate so we can use the same on APM / PX4 / Solo in the medium term.

Let me know if the intent for some of the fields is not entirely clear.

```
Follow-Me message:

uint64_t timestamp
uint32_t lat (deg * 1e7)
uint32_t lon (deg * 1e7)
float alt (AMSL, meters)
float velocity[3]; (0 0 0 for unknown)
float accel[3]; (0 0 0 for unknown)
float attitude_q[4]; (1 0 0 0 for unknown)
float rates[3]; (0 0 0 for unknown)
float pos_std_dev[3]; -1 for unknown
uint8_t est_capabilities; (POS, VEL, ACCEL, ATT + RATES)
uint64_t custom_state;
```

liamstask, 2016-02-15:
/cc @ne0fhyk who has worked on follow me GCS functionality

jschall, 2016-02-16:
The intent definitely needs to be fleshed out for those fields. It isn't even clear whether these are intended to be target states or vehicle demands.

Essentially what is needed is position of the follow target and derivatives if desired, and extra points for an NED position, velocity, acceleration demand for the vehicle in a moving coordinate system where the follow target is at the origin. Don't at all get what the quaternion is for.

catch-twenty-two, 2016-02-16:
What about target system/component?

rmackay9, 2016-02-17:
Yes, I agree with @jschall.  There's should be a message to say what system-id/component-id the vehicle should follow and then at least a position offset and possibly some other way of describing that we want the vehicle to orbit the target.. i'm not sure how we would express that.. perhaps a 3D vector of rotation rates about the target or perhaps velocities (i.e. tangential where Z is towards the target)?  

Then the  target is responsible for spitting out it's position, velocity, etc using the regular http://mavlink.org/messages/common#GLOBAL_POSITION_INT message (or similar).

LorenzMeier, 2016-02-17:
Let me take another step and try to explain why I think GLOBAL_POSITION_INT is not good enough. I also factored in the missing commands you raised which I indeed missed.

Note that I'm currently not building a device / product with this - this is solely driven by the desire to add this to QGC running on a tablet. So parts of it are over-engineered, but driven by what I have seen in commercial products, ranging from Airdog to Solo.

```C
Follow-Me message:
MAVLINK_MSG_ID_FOLLOW_TARGET
// timestamp, UNIX epoch (GPS synced)
uint64_t timestamp
uint8_t est_capabilities
// Flags to indicate which quantities the tracker knows about: (POS, VEL, ACCEL, ATT + RATES)
// target position
uint32_t lat (deg * 1e7)
uint32_t lon (deg * 1e7)
float alt (AMSL, meters)
// target velocity
float velocity[3]; (0 0 0 for unknown)
// target acceleration (linear) in earth frame. This is powerful to predict motion, in particular in action sports
float accel[3]; (0 0 0 for unknown)
// where the target is facing. This will be often ignored, but if you mount a good tracker to a helmet you can make sure to get a shot including of the face of the person. Also powerful for larger objects like cars and boats where the shot should be relative to the vehicle.
float attitude_q[4]; (1 0 0 0 for unknown)
float rates[3]; (0 0 0 for unknown)
// uncertainly in earth frame for X, Y and Z. We will need to agree on the exact format here
float pos_cov[3]; -1 for unknown
// A custom vector, can be used to transmit e.g. button states or switches of a tracker device
uint64_t custom_state;
```

```C
Follow-Me command:
MAV_CMD_DO_FOLLOW
param1: System ID (the system ID of the FOLLOW_TARGET beacon). Send 0 to disable follow-me and return to the default position hold mode
param2: distance to target
param3: altitude flag: 0: Track at altitude relative to target, 1: Track at fixed altitude, 2: Track based on current alt difference, 3: Track based on current absolute altitude
param4: angle / mode flag: 0: The angle is relative to the earth fixed motion vector (so the drone will follow the trajectory of the target at the specified angle. 1: The angle is relative to the heading of the target (for a boat or truck, so the camera can always stay in front)
param5: altitude
param6: angle relative to target, should 0 be behind target or in front?
```

lvale, 2016-02-17:
I would add a TTL that would define the expected intervals between updated positions. If the vehicle doesn't receive an update within the TTL then should enter Loiter.

LorenzMeier, 2016-02-17:
makes a lot of sense

lvale, 2016-02-17:
I'm not sure if Loiter only is good enough, perhaps Loiter defaulting to either AltHold or Land.

rmackay9, 2016-02-18:
I think a 3D position offset from the target would be better than a distance and angle.
I don't see how we'd capture the orbit request with this.  We would need a rotation about the target field in 3D.

LorenzMeier, 2016-02-18:
I reworked my post above:
https://github.com/mavlink/mavlink/issues/509#issuecomment-185314193

catch-twenty-two, 2016-03-17:
Are we good with this structure for now?  In order to pull my changes for follow target into QGC and the PX4 base it's my understanding that the message needs to be in an auto generated branch that QGC and px4 can point to in their submodules. So before we can get the px4 and qgc branches integrated (which are ready to be tested) I'll need to get a merge with this new message. Let me know and Ill create a new pr with the agreed upon structure above.

catch-twenty-two, 2016-03-18:
New PR here:
https://github.com/mavlink/mavlink/pull/526

LorenzMeier, 2016-04-19:
Closing as its merged - please refer to the added message for further use. Let's not lock it down yet until we have seen more validation and the APM dev team has started to use it and is happy with its fields.

closed 2016-04-19 by LorenzMeier

tridge, 2016-05-30:
I've been working on a project that involves a followme type task (landing on a moving boat with a quadplane). One thing I noticed is that the existing messages don't offer any way to do time sync. Without time synchronisation the vehicle can't extrapolate the desired position using the velocity with any accuracy.
I'm thinking of adding yet another message called SET_LAND_TARGET for my application, which would have GPS week and GPS tow in milliseconds, as that offers a common timebase that doesn't suffer from problem like UTC leap second offsets.

tridge, 2016-05-30:
I'd also note that FOLLOW_TARGET uses float[3] arrays, which aren't supported in pymavlink. We could extend pymavlink to understand arrays of other than bytes, but it is quite a tricky patch.