mavlink/mavlink #510: Common.xml: Wind message

LorenzMeier, 2016-02-17:
@magicrub @tridge Driven by the interest for common implementations driven by the full APM/PX4 support in QGroundControl I'd like to come up with a Wind message for common.xml. Would you want to add fields to what you're using right now? Should we try to introduce height layers? The amount of payload data vs. the amount of header bytes is for this message relatively low.

```xml
        <message id="168" name="WIND">
            <description>Wind estimation</description>
            <field name="direction" type="float">wind direction that wind is coming from (degrees)</field>
            <field name="speed" type="float">wind speed in ground plane (m/s)</field>
            <field name="speed_z" type="float">vertical wind speed (m/s)</field>
        </message>
```

LorenzMeier, 2016-02-17:
@priseborough @tumbili @AndreasAntener @SimonWilks Please chime in!

Susurrus, 2016-03-29:
This could be made 50% smaller if everything was in units of 0.01 and datatypes were all uint16.

Additionally the range of the `direction` field should be specified, I suggest 0..2pi, as most fields are specified this way.

LorenzMeier, 2016-04-19:
Most fields are specified -pi..pi, not 0 to pi, which trigonometrically doesn't actually work.

LorenzMeier, 2016-04-19:
Closing in favour of https://github.com/mavlink/mavlink/pull/542/files

closed 2016-04-19 by LorenzMeier