mavlink/mavlink #511: Common.xml: Estimator status report

LorenzMeier, 2016-02-17:
@magicrub @tridge Same as for wind: It would be great to introduce a common message for QGC's use (it will support the APM version and the common version in a transitional time which can be quite extended). What would you like to see changed?

```xml
        <message id="xxx" name="EKF_STATUS_REPORT">
            <description>EKF Status message including flags and variances</description>
            <field name="flags" type="uint16_t">Flags</field>
            <!-- supported flags see EKF_STATUS_FLAGS enum -->
            <field name="velocity_variance" type="float">Velocity variance</field>
            <!-- below 0.5 is good, 0.5~0.79 is warning, 0.8 or higher is bad -->
            <field name="pos_horiz_variance" type="float">Horizontal Position variance</field>
            <field name="pos_vert_variance" type="float">Vertical Position variance</field>
            <field name="compass_variance" type="float">Compass variance</field>
            <field name="terrain_alt_variance" type="float">Terrain Altitude variance</field>
        </message>
```

Status flags:

```xml
        <!-- EKF_STATUS_FLAGS - these values should be bit-and with the messages flags field to know if flag has been set -->
        <enum name="EKF_STATUS_FLAGS">
            <description>Flags in EKF_STATUS message</description>
            <entry name="EKF_ATTITUDE" value="1">
                <description>set if EKF's attitude estimate is good</description>
            </entry>

            <entry name="EKF_VELOCITY_HORIZ" value="2">
                <description>set if EKF's horizontal velocity estimate is good</description>
            </entry>

            <entry name="EKF_VELOCITY_VERT" value="4">
                <description>set if EKF's vertical velocity estimate is good</description>
            </entry>

            <entry name="EKF_POS_HORIZ_REL" value="8">
                <description>set if EKF's horizontal position (relative) estimate is good</description>
            </entry>

            <entry name="EKF_POS_HORIZ_ABS" value="16">
                <description>set if EKF's horizontal position (absolute) estimate is good</description>
            </entry>

            <entry name="EKF_POS_VERT_ABS" value="32">
                <description>set if EKF's vertical position (absolute) estimate is good</description>
            </entry>

            <entry name="EKF_POS_VERT_AGL" value="64">
                <description>set if EKF's vertical position (above ground) estimate is good</description>
            </entry>

            <entry name="EKF_CONST_POS_MODE" value="128">
                <description>EKF is in constant position mode and does not know it's absolute or relative position</description>
            </entry>

            <entry name="EKF_PRED_POS_HORIZ_REL" value="256">
                <description>set if EKF's predicted horizontal position (relative) estimate is good</description>
            </entry>

            <entry name="EKF_PRED_POS_HORIZ_ABS" value="512">
                <description>set if EKF's predicted horizontal position (absolute) estimate is good</description>
            </entry>
        </enum>
```

LorenzMeier, 2016-02-17:
@priseborough it would be great if you could join in, along with @tumbili.

LorenzMeier, 2016-02-17:
I would propose to add the vibration levels here. We've stuck them in PX4 into the same data structure as the estimator core status.

Susurrus, 2016-02-18:
> no new ideas here, looks good.

Any reason why these can't be fixed point? The proliferation of floating point datatypes in MAVLink messages is problematic for many lower-end MCUs and I think they should be avoided if at all possible because of this.

priseborough, 2016-02-18:
I would like to see the descriptions for the normalised innovations (called variances) updated and expanded to clarify that they represent the ratio of the respective innovations fail thresholds. I'm not sure what 'terrain_alt_variance'  is supposed to be, I assume it is the normalised range finder innovation. We also need to add the normalised innovation for the airspeed measurements.

LorenzMeier, 2016-02-18:
Would you mind just doing the changes, either editing my post or posting your version below? That's probably the easiest.

priseborough, 2016-02-18:
Can you send me the link to the source code - My browser window cuts off some of the text in your post

priseborough, 2016-02-24:
<message id="xxx" name="EKF_STATUS_REPORT">
            <description>EKF Status message including flags, innovation test ratios and estimated accuracies. The flags message is an integer bitmask containing information on which EKF outputs are valid. See the EKF_STATUS_FLAGS enum definition for further information. The innovaton test ratios show the magnitude of the sensor innovation divided by the innovation check threshold. Under normal operation the innovaton test ratios should be below 0.5 with occasional values up to 1.0. Values greater than 1.0 should be rare under normal operation and indicate that a measurement has been rejected by the filter. The user should be notified if an innovation test ratio greater than 1.0 is recorded. Notifications for values in the range between 0.5 and 1.0 should be optional and controllable by the user.</description>
            <field name="flags" type="uint16_t" enum="EKF_STATUS_FLAGS">Integer bitmask indicating which EKF outputs are valid. See definition for EKF_STATUS_FLAGS.</field>
            <field name="vel_ratio" type="float">Velocity innovation test ratio</field>
            <field name="pos_horiz_ratio" type="float">Horizontal position innovation test ratio</field>
            <field name="pos_vert_ratio" type="float">Vertical position innovation test ratio</field>
            <field name="mag_ratio" type="float">Magnetometer innovation test ratio</field>
            <field name="HAGL_ratio" type="float">Height above terrain innovation test ratio</field>
            <field name="TAS_ratio" type="float">True airspeed innovation test ratio</field>
            <field name="POS_horiz_accuracy" type="float">Horizontal position 1-STD accuracy relative to the EKF local origin (m)</field>
            <field name="POS_vert_accuracy" type="float">Vertical position 1-STD accuracy relative to the EKF local origin (m)</field>
        </message>

        
        <enum name="EKF_STATUS_FLAGS">
            <description>Flags in EKF_STATUS message</description>
            <entry name="EKF_ATTITUDE" value="1">
                <description>True if the attitude estimate is good</description>
            </entry>

            <entry name="EKF_VELOCITY_HORIZ" value="2">
                <description>True if the horizontal velocity estimate is good</description>
            </entry>

            <entry name="EKF_VELOCITY_VERT" value="4">
                <description>True if the  vertical velocity estimate is good</description>
            </entry>

            <entry name="EKF_POS_HORIZ_REL" value="8">
                <description>True if the horizontal position (relative) estimate is good</description>
            </entry>

            <entry name="EKF_POS_HORIZ_ABS" value="16">
                <description>True if the horizontal position (absolute) estimate is good</description>
            </entry>

            <entry name="EKF_POS_VERT_ABS" value="32">
                <description>True if the vertical position (absolute) estimate is good</description>
            </entry>

            <entry name="EKF_POS_VERT_AGL" value="64">
                <description>True if the vertical position (above ground) estimate is good</description>
            </entry>

            <entry name="EKF_CONST_POS_MODE" value="128">
                <description>True if the EKF is in a constant position mode and is not using external measurements (eg GPS or optical flow)</description>
            </entry>

            <entry name="EKF_PRED_POS_HORIZ_REL" value="256">
                <description>True if the EKF has sufficient data to enter a mode that will provide a (relative) position estimate</description>
            </entry>

            <entry name="EKF_PRED_POS_HORIZ_ABS" value="512">
                <description>True if the EKF has sufficient data to enter a mode that will provide a (absolute) position estimate</description>
            </entry>

priseborough, 2016-02-24:
I've reworked the descriptions, added the missing airspeed innovation test ratio and added a couple of fields for position accuracy. We can probably rework these to use int16's and save some bandwidth.

LorenzMeier, 2016-03-27:
It just came up that we don't have a very good reporting path for the GPS input accuracy. Should we report this here as well, or rather create a new GPS message?

julianoes, 2016-03-27:
I vote to add the eph and epv values reported by uBlox GPS to this message since they are not included in the gps_raw_int message (anymore).

priseborough, 2016-03-28:
This message is about the health of the estimator. If an accuracy is to be reported in this message, it should be the estimators eph and epv (calculated from state variances).

Why don't we report the raw GPS eph and epv in gps_raw_int?

julianoes, 2016-03-29:
@priseborough see discussion here: https://github.com/mavlink/mavlink/issues/527

LorenzMeier, 2016-04-19:
Closing in favour of https://github.com/mavlink/mavlink/pull/542/files. The position accuracy has been captured there, and from a user perspective (and given we get flow in the loop) accuracy in meters in 1 std is more meaningful for the overall estimator.

closed 2016-04-19 by LorenzMeier