mavlink/mavlink #514: GPL license in several pymavlink files

cajlarsson, 2016-02-24:
It seems not all files had their license header changed to LGPLv3 and stayed GPLv3. Which I assume makes this whole lib GPLv3. I'm sure this is not intentional and all cases had a header that looked something like

    '''
    mavlink python utility functions
    Copyright Andrew Tridgell 2011
    Released under GNU GPL version 3 or later
    '''

cajlarsson, 2016-02-24:
I'm blind

magicrub, 2016-02-24:
Not blind?

cajlarsson, 2016-02-24:
The blindness seems to have only been temporary!

peterbarker, 2016-09-05:
@cajlarsson On what basis did you believe the pymavlink files should have been relicensed?

I know there was an issue concerning the products of any generation that might occur - an issue to clarify that is here: https://github.com/mavlink/mavlink/issues/549

AFAIK, nobody is seeking to change the licenses within e.g. mavutil.py!

Can this issue be closed?

cajlarsson, 2016-09-15:
Yes, that is completely verifying for me, thank you!

magicrub, 2016-09-15:
Ready to close?

LorenzMeier, 2016-09-15:
Yes - we never discussed or welcomed to relicense the generator / pyMAVLink. The discussion was only ever about clarifying the output.

closed 2016-09-15 by LorenzMeier