mavlink/mavlink #525: MAV_CMD_WAYPOINT_USER_1 is not displayed correctly

Fojtik, 2016-03-16:
Dears, before while we agreed that new special waypoints types will be added. These waypoints might have custom payload and QGC wil display them transparently.

Unfortunatelly, when I test to use MAV_CMD_WAYPOINT_USER_1, QGC does not display its position.

DonLakeFlyer, 2016-03-22:
The mavlink pull hasn't not been merged in yet so QGC has not been updated to support this yet.

Fojtik, 2016-03-22:
OK, thank you for info - are the QGC developpers merging sometimes, or where to ask to merge this? Is this github right platform for merge to ask?

DonLakeFlyer, 2016-03-22:
I have it on my list of todos. I’ll work to get this pushed through.

closed 2016-03-23 by DonLakeFlyer

Fojtik, 2016-03-26:
It unfortunatelly does not work, see attached image.
[image]

Fojtik, 2016-10-09:
Sorry, custom commands still does not work even in version 3.0.1.

magicrub, 2016-10-09:
The command should be there, the problem of not working is a px4/ardupilot problem. They're not implemented yet

Fojtik, 2016-10-09:
To avoid duplicity in discussion I recommend to continue in [https://github.com/mavlink/mavlink/issues/633#issuecomment-252511318](https://github.com/mavlink/mavlink/issues/633#issuecomment-252511318)