mavlink/mavlink #527: Add HDOP VDOP fields to gps_raw_int packet

jgoppert, 2016-03-21:
Any reason not to add hdop and vdop fields and keep eph and epv since the standard deviation of the error is a useful number.

```bash
          <field type="uint16_t" name="eph">GPS HDOP horizontal dilution of position (unitless). If unknown, set to: UINT16_MAX</field>
               <field type="uint16_t" name="epv">GPS VDOP vertical dilution of position (unitless). If unknown, set to: UINT16_MAX</field>
```

LorenzMeier, 2016-03-23:
Would break the message format, not significant enough to justify the incompatibility ramifications.

LorenzMeier, 2016-03-26:
Will not happen.

julianoes, 2016-03-26:
Sorry, I gotta re-open this. It got consfused today as well. Either the comment or the message needs to be changed.

Personally, I'd vote to have eph/epv and hdop/vdop in the message because GPS signal is important enough to know enough about it while planning and carrying out flights.

If we need to decide between eph/epv vs hdop/vdop, I vote for eph because there is an actual unit (m) to the number which is intuitive. When trying to assess if the GPS signal is "good" enough, it is much more helpful to look at eph/epv (at least for uBlox) because the reported GPS position can be rubbish even if the hdop is "good".

Today for instance, I read an hdop value of 1.2 which is excellent according to [wikipedia](https://en.wikipedia.org/wiki/Dilution_of_precision_(GPS)#Meaning_of_DOP_Values) and at the same time position drifts were exceeding 100m. Presumably this is due to interference which is not taken into account by hdop/vdop.
In the past, I made the experience that eph/epv are a good indicator whether we should trust the signal or not, and it helps to debug interference issues.

LorenzMeier, 2016-03-27:
We can add that to the estimator status packet. But we will not make the GPS status packet *UNREADABLE* (which adding fields will do).

HDOP has been consistent except for your case, so I'm presuming its some sort of corner case. We could only consider adding the fields for a new GPS message.

Issue for estimator status reporting:
https://github.com/mavlink/mavlink/issues/511

closed 2016-03-27 by LorenzMeier

julianoes, 2016-03-27:
> We can add that to the estimator status packet. But we will not make the GPS status packet UNREADABLE (which adding fields will do).

Ok, let's add it to the estimator status packet. I understand that we can't break the current field. However, can we make sure to get this right in a later revision?

> HDOP has been consistent except for your case, so I'm presuming its some sort of corner case. We could only consider adding the fields for a new GPS message.

I strongly disagree. I think interference is more than a "corner case".
As said above, hdop is only based on the geometry of the available satellites and therefore no indicator of actual accuracy of the position/velocity estimate. It can still be a helpful information though, sure.