mavlink/mavlink #529: Java: uint64_t is not handled correctly

WickedShell, 2016-03-22:
Due to java not having unsigned types we need to promote one size whenever we encounter a unsigned type. Unfortunately there are no primitive types larger then a long, so to handle a long we would need to shove it into a BigInteger, which should be fine as that is part of java.Math and has been around since JDK 1.1.

Does anyone have a concern about using BigIntegers to encode any uint64_t's?

LorenzMeier, 2018-05-26:
Closing as stale.

closed 2018-05-26 by LorenzMeier

WickedShell, 2018-05-26:
It's not stale. Our should probably be moved to the pymavlink issue, but the issue itself is most definitely not stale.

I do not understand closing issues out that are still issues just because no one has commented on it recently.

hamishwillee, 2018-05-28:
> I do not understand closing issues out that are still issues just because no one has commented on it recently.

I'm not a fan either. Lets see if there is a policy: #925

In the meantime, could you please move this issue to the right place?