mavlink/mavlink #540: Request: Loiter Until Command (or Loiter System Time)

labtoast, 2016-04-14:
I need two vehicles to exit a loiter at the exact same time without external inputs.

I propose a loiter until command that causes vehicle to wait until `system time` has elapsed commanded time. This would be used for synchronizing a patrol path or synchronizing movements between multiple vehicles in an open loop manner. 

Inputs:
- Timestamp
- Relative/Absolute Time
- Standard Loiter Parameters

```
<entry value="19" name="MAV_CMD_NAV_LOITER_UNTIL">
                    <description>Loiter around this MISSION until X time</description>
                    <param index="1">Seconds (decimal)</param>
                    <param index="2">0 - time since boot; 1 - absolute time</param>
                    <param index="3">Radius around MISSION, in meters. If positive loiter clockwise, else counter-clockwise</param>
                    <param index="4">Desired yaw angle.</param>
                    <param index="5">Latitude</param>
                    <param index="6">Longitude</param>
                    <param index="7">Altitude</param>
               </entry>
```

Previously requested here https://github.com/mavlink/mavlink/issues/440
Request: Loiter Until Command #440

magicrub, 2016-04-14:
Please refer to the following:
issue https://github.com/ArduPilot/ardupilot/issues/3675
issue https://github.com/ArduPilot/ardupilot/issues/939
PR https://github.com/ArduPilot/ardupilot/pull/3708
video: https://www.youtube.com/watch?v=9VK3yjIyCSo

Turns out this can get a little more complicated once you start considering a few things.

labtoast, 2016-04-21:
Thanks for the info! I'll close this due to the related issues.

closed 2016-04-21 by labtoast