mavlink/mavlink #544: Unable to translate payload data with C++

harewei, 2016-04-22:
I'm currently trying to write a C++ program that reads mavlink messages
I've built the library with mavlink generator, and I can get the mavlink message through USB port.
I can access things like sysid, msgid, messageInfo[msg.msgid].fields[i].wire_offset etc
The only problem left is I don't know how to interpret the payload data

```
int res = read(fd,buf,BUFFER_LENGTH);
if (res > 0)
      	{
			// Something received
			mavlink_message_t msg;
			mavlink_status_t status;

for (i = 0; i < res; ++i)
			{
				temp = buf[i];
				//printf("%02x ", (unsigned char)temp);	// Print at least 2 digits, prepend it with 0 's if there's less
				if (mavlink_parse_char(MAVLINK_COMM_0, buf[i], &msg, &status))
				{
					// Packet received
					//printf("\nReceived packet: SYS: %d, COMP: %d, LEN: %d, MSG ID: %d\n", msg.sysid, msg.compid, msg.len, msg.msgid);
				
					mavlink_message_info_t messageInfo[256] = MAVLINK_MESSAGE_INFO;

// Decide what to do for each of the field
					for (unsigned int i = 0; i < messageInfo[msg.msgid].num_fields; ++i)
					{
						cout << "field name: " << messageInfo[msg.msgid].fields[i].name << endl;
						switch (messageInfo[msg.msgid].fields[i].type)
						{
							case MAVLINK_TYPE_FLOAT:
							//cout << "got float\n\n\n\n\n" << endl;
								if (messageInfo[msg.msgid].fields[i].array_length > 0)
								{									
									
								}
								else
								{
									// Single value
									printf("msgid %d \n", msg.msgid);
									cout << "wireoffset: " << messageInfo[msg.msgid].fields[i].wire_offset << endl;
									//float f = buf[8+messageInfo[msg.msgid].fields[i].wire_offset];
									float f;
									unsigned int b[] = {buf[messageInfo[msg.msgid].fields[i].wire_offset], 
										buf[messageInfo[msg.msgid].fields[i].wire_offset+1], 
										buf[messageInfo[msg.msgid].fields[i].wire_offset+2], 
										buf[messageInfo[msg.msgid].fields[i].wire_offset+3]};
									memcpy(&f, &b, sizeof(f));
									cout << "float: " << f << endl;
								}
								break;
```

I'm trying to read the float value as you can see in my code, and I thought it would be stored in buf array at location of each field wire_offset with 4 bytes (which combines into a float).  Unfortunately I'm not getting the correct values.  Can someone help me with this?  Thanks.

magicrub, 2016-04-22:
you've posted a help question in the developers bug tracker TO-DO list and/or feature request system. Please close and ask on a forum or on gitter, try here: https://gitter.im/mavlink/DocDiscussion

amilcarlucas, 2018-02-19:
This can be closed now

closed 2018-02-20 by hamishwillee