mavlink/mavlink #549: License clarification of generated MAVLink files

LorenzMeier, 2016-04-28:
I did get approached by companies (startups, not corporate) which were concerned that their use of the MAVLink library in commercial products could be an issue. While our intent is clearly to maximise the proliferation of the protocol, we also want to retain the generator and infrastructure under a copyleft license system.

@tidge and I discussed this and are proposing a model similar to GCC: The "compiler" is GPL licensed, while the resulting compiler output is permissive. This will ensure that improvements to the protocol remain open but it can be bundled into any sort of system. This is in particular relevant for embedded systems.

We would like to ask every pinged author to:

  - Approve or reject the licensing of the *generated* files as MIT
  - When approving clearly indicate that the code used in the templates themselves and utility functions is does not contain third party code which is not MIT compatible (e.g. GPL)

The proposed extension of the license text is:

```
## Exception to the LGPL v.3: ##
As an exception, if you use this Software to compile your source code and
portions of this Software, including messages ("the generator output"), are embedded into the
binary product as a result, you may redistribute such product and such
product is hereby licensed under the following MIT license:

Permission is hereby granted, free of charge, to any person obtaining a copy
of this software and associated documentation files (the "Software"), to deal
in the Software without restriction, including without limitation the rights
to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the Software is
furnished to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in
all copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT.  IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
THE SOFTWARE.
```

LorenzMeier, 2016-04-28:
@jgoppert @Susurrus @liamstask

tridge, 2016-04-28:
I think the 2nd para needs to specifically refer to the generated code, otherwise it could be ambiguous if it refers to the generator or the output. The first time I read it I thought "the Software" meant the mavlink package overall. On 2nd reading it was clearer. I'd propose the following to make it clear on first reading:

  Permission is hereby granted, free of charge, to any person obtaining a copy
  of the generated software (the "Generated Software"), to deal
  in the Generated Software without restriction, including without limitation the rights
  to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
  copies of the Software, and to permit persons to whom the Generated Software is
  furnished to do so, subject to the following conditions:

note that I also removed the "and associated documentation files" as I'm not actually sure which docs we are referring to there?

LorenzMeier, 2016-04-28:
Ok, I copied it 1:1 into my original post, so that's now the final proposal.

tridge, 2016-04-28:
I just realised that the references to "the Software" in the later paragraphs also should be changed to "the Generated Software", otherwise the definition doesn't apply

jgoppert, 2016-04-28:
I approve and did not use any non MIT compatible code to my knowledge.

liamstask, 2016-04-28:
Same, I approve the licensing of the generated files as MIT, and did not use any third party code which is not MIT compatible.

LorenzMeier, 2016-04-29:
@barthess @thomasgubler @dagoodma @geeksville @vooon @dimasad @wiml @malife @rlhartley @ryancu7 @jlecoeur Please let us know each here if you approve the licensing of the generated files as MIT and wether or not you used non-MIT compliant code in your contributions. Thanks!

LorenzMeier, 2016-04-29:
This would conclude the authors of the C generator files.

ryancu7, 2016-04-29:
I approve the licensing of the generated files as MIT, and did not use any third party code which is not MIT compatible.

jlecoeur, 2016-04-29:
I approve the licensing of the generated files as MIT, and did not use any third party code which is not MIT compatible.

vooon, 2016-04-29:
I approve the licensing of the generated files as MIT, and did not use any third party code which is not MIT compatible.

dimasad, 2016-04-29:
I strongly approve and haven't used anything that is incompatible with MIT.
In fact, I can't imagine anything that is incompatible with MIT, which is a
great advantage of using permissive licensing for anyone who doesn't have a
law degree. Two simple clauses which are very clear and simple to comply.

~Dimas

barthess, 2016-04-29:
I approve the licensing of the generated files as MIT, and did not use any third party code which is not MIT compatible.

malife, 2016-04-29:
I approve the licensing of the generated files as MIT, and did not use any third party code which is not MIT compatible.

-- 
Mariano Lizarraga

Reply: mavlink/mavlink <reply@reply.github.com>
Date: April 29, 2016 at 3:10:42 AM
To: mavlink/mavlink <mavlink@noreply.github.com>
CC: Mariano Lizarraga <malife@gmail.com>, Mention <mention@noreply.github.com>

geeksville, 2016-04-29:
Fine by me.  I didn't add any third party code and I'm okay with the stuff I wrote being MIT licensed for generated files.

Susurrus, 2016-04-29:
This is alright by me as well. I haven't used any third-party code not compatible with the MIT license.

LorenzMeier, 2016-04-29:
@brucecrevensten Would also be great to have your feedback.

thomasgubler, 2016-04-30:
I approve the licensing of the generated files as MIT, and did not use any third party code which is not MIT compatible.

LorenzMeier, 2016-05-02:
@dagoodma @vooon @wiml Would be great to get your feedback as well.

dagoodma, 2016-05-06:
Lorenz,

Fine by me. I approve the licensing of the generated files as MIT, and to
my knowledge did not use any third party code which is not MIT compatible.

ne0fhyk, 2016-05-20:
I approve the licensing of the generated files as MIT, and did not use any third party code which is not MIT compatible.

arthurbenemann, 2016-05-22:
+1 for the Java generator

closed 2016-09-15 by LorenzMeier