mavlink/mavlink #553: mavlink ROS package no longer includes pymavlink

cberner, 2016-05-15:
The latest ROS packages for indigo and jade don't seem to include pymavlink anymore, even though the wiki states that they do: http://wiki.ros.org/mavlink

vooon, 2016-05-15:
Oh, again... I checked [ros-jade-mavlink_2016.4.4-0trusty-20160403-172613-0700_i386.deb](http://packages.ros.org/ros-shadow-fixed/ubuntu/pool/main/r/ros-jade-mavlink/ros-jade-mavlink_2016.4.4-0trusty-20160403-172613-0700_i386.deb) - headers only.

Interesting thing that locally from source pymavlink was build.

vooon, 2016-05-15:
Ok, i found trace back:
http://build.ros.org/view/Jbin_uV32/job/Jbin_uV32__mavlink__ubuntu_vivid_i386__binary/5/console
```
18:03:56 + [ -n /tmp/binarydeb/ros-jade-mavlink-2016.4.4/debian/ros-jade-mavlink ]
18:03:56 + DESTDIR_ARG=--root=/tmp/binarydeb/ros-jade-mavlink-2016.4.4/debian/ros-jade-mavlink
18:03:56 + /usr/bin/env /usr/bin/python /tmp/binarydeb/ros-jade-mavlink-2016.4.4/pymavlink/setup.py build --build-base /tmp/binarydeb/ros-jade-mavlink-2016.4.4/obj-i686-linux-gnu/pymavlink/pybuild install --root=/tmp/binarydeb/ros-jade-mavlink-2016.4.4/debian/ros-jade-mavlink --install-layout=deb --prefix=/opt/ros/jade
18:03:56 Traceback (most recent call last):
18:03:56   File "/tmp/binarydeb/ros-jade-mavlink-2016.4.4/pymavlink/setup.py", line 11, in <module>
18:03:56     from setuptools import setup, Extension
18:03:56 ImportError: No module named setuptools
```

vooon, 2016-05-15:
I added setuptools and released:

- Indigo: https://github.com/ros/rosdistro/pull/11518
- Jade: https://github.com/ros/rosdistro/pull/11520
- Kinetic: https://github.com/ros/rosdistro/pull/11521

vooon, 2016-05-15:
From Jenkins i see issue is fixed, but i made small changes to failure on pymavlink install error:

- Indigo: https://github.com/ros/rosdistro/pull/11522
- Jade: https://github.com/ros/rosdistro/pull/11523
- Kinetic: https://github.com/ros/rosdistro/pull/11524

vooon, 2016-05-16:
Fresh deb's have pymavlink. Fixed.

http://packages.ros.org/ros-shadow-fixed/ubuntu/pool/main/r/ros-jade-mavlink/

closed 2016-05-16 by vooon