mavlink/mavlink #566: Unable to generate code files

BrianCohen, 2016-06-06:
I have Python and the cloned repository folder set in the environmental variables Path. But when I try to run commands like python -m mavgenerate or python -m pymavlink.tools.mavgen, it says:
`C:\Python27\python.exe: No module named pymavlink.tools`

Thanks in advance

magicrub, 2016-06-06:
Have you tried mavlink/pymavlink/tools/mavgen.py?

BrianCohen, 2016-06-06:
Do you mean `python -m mavlink\pymavlink\tools\mavgen.py` ?
It says ` Import by filename is not supported.`
If I take out the .py its the same as before.

It seem like it's not getting the python path correctly? 
By the way I'm using windows 10 and have tried with Python 2.7 and Python 3.5

magicrub, 2016-06-06:
Which project are you compiling this for? If it's ArduPilot this is all done for you during make

BrianCohen, 2016-06-06:
No, I'm trying to generate the language-specific source files to use in my own c# project.

BrianCohen, 2016-06-07:
Ok, I've been able to generate files using `python mavgenerate.py` and using mavgen.py with the whole pathname because otherwise it can't find it. But then it does something weird. 
It exits the script saying it cant' locate `CS\common\ByteArrayUtil.cs`  and two other files. But it generates two files, `CS.generated.cs` and `CS_codec.generated` , CS being the name of the output directory. So it doesn't create a directory but uses as prefix for the files.

Is this right? Are those the only files that are supposed to be generated for C# ?

Thanks

BrianCohen, 2016-06-13:
Anyone can confirm if the generated files are all right? 
It outputs three "source file not found" error:
`error CS2001: No se encuentra el archivo de código fuente 'CS\common\ByteArrayUtil.cs'
error CS2001: No se encuentra el archivo de código fuente 'CS\common\FrameworkBitConverter.cs'
error CS2001: No se encuentra el archivo de código fuente 'CS\common\Mavlink.cs'
Error`
But the python window says that messages are generated successfully.

amilcarlucas, 2018-02-21:
AFAIK this is a duplicate of https://github.com/ArduPilot/pymavlink/issues/151 and can be closed.

amilcarlucas, 2019-04-05:
No answer in over three years

closed 2019-04-05 by amilcarlucas