mavlink/mavlink #570: common.xml message with id 256

dsuarezv, 2016-06-12:
Hi, a recent commit (184893c0b9f6d95a05d4af29a809762e4614a224) has introduced a new message in common.xml with id 256. 

I understand that the message id field is supposed to be a byte, at least in http://qgroundcontrol.org/mavlink/create_new_mavlink_message, it says 

> id=“0” means this message has the id / index number zero. 
> Valid numbers range from 0 to 255, with ids 150-240 being reserved for extensions (so you will want to use the 150-240 range for your personal messages)

Is this a mistake or rules have changed regarding the message id field?

vooon, 2016-06-12:
It is for MAVLink v2.0, which has 24-bit message id. And as i know only C and Python implementations currently support 2.0 features.

dsuarezv, 2016-06-13:
Looks like a breaking change, shouldn't a message like this go into a new folder "v2.0"?

vooon, 2016-06-13:
V2 library may produce and receive V1 messages with ID < 256. So i think @tridge decided not to copy xml's.

Also V2 has message signing and variable wire message length (trim padding zero bytes).

tridge, 2016-06-13:
@dsuarezv,
we just need to fix the cs generator to handle message IDs above 256 just like the C and python one does.
If I make a patch for the cs generator are you able to test it for me?

dsuarezv, 2016-06-13:
Actually I'm a developer of https://github.com/dsuarezv/mavlink.net, it's another generator for C#, so we only use the xml files to generate a different set of classes. 

Am I right thinking the proper way to handle this in a V1 generator would be to simply ignore messages with id > 255?

magicrub, 2016-06-13:
I would think so.

dsuarezv, 2016-06-13:
Ok, will do that to quick-fix the issue while we work on the V2 implementation. 

The best reference for V2 is the C generator, right?

magicrub, 2016-06-13:
yes, C gen is reference.

tridge, 2016-06-13:
there is also some documentation here:
https://github.com/mavlink/mavlink/blob/master/doc/MAVLink2.md

tridge, 2016-06-13:
also, would you like to keep your generator separate or would you be interested in folding it into the mainline mavlink? I don't know how many people are using the current cs generator. If there are users then we could have both generators, otherwise if there aren't any users we could replace the current one with yours

dsuarezv, 2016-06-13:
I think integrating it in the mainline mavlink is the natural place for it to be. Keep in mind, however, that it is a separate project written entirely in c#, maybe it stands out from the other generators (different usage, different build deps), and it is not API compatible with your current generator (I think that at least MissionPlanner uses these bindings, right?)

magicrub, 2016-06-13:
Tridge, please check with Mission planner first.

hamishwillee, 2020-04-23:
Closing. I believe this was addressed

closed 2020-04-23 by hamishwillee