mavlink/mavlink #592: MAVLink with the Pixhawk some commands not working.

ruralguru, 2016-07-14:
I am utilizing an off-board computer to control the Pixhawk using v3.3.3 firmware
The MAVLink documentation states that REQUEST_DATA_STREAM is deprecated and to instead use SET_MESSAGE_INTERVAL.
However when I try to utilize the command_long encode to send I get no response as per the following c++ code:

**mavlink_command_long_t com = { 0 };
com.target_system    = system_id;
com.target_component = autopilot_id;
com.command          = MAV_CMD_GET_MESSAGE_INTERVAL;
com.confirmation     = 0;
com.param1           = mav_msg_ID; //The MAVLink message ID
com.param2						= interval_us; //The interval between two messages, in microseconds. Set to -1 to disable and 0 to request default rate.
printf("Sent request for ID#:%.0f to be output every %.0f us\n",com.param1, com.param2);

mavlink_message_t message;
mavlink_msg_command_long_encode(system_id, companion_id, &message, &com);	 //Encode

int len = serial_port->write_message(message); //Send the message
return len;**

this is based off the ned write utilized in c_uart_example code also under mavlink.

ruralguru, 2016-07-14:
oops wrong thread

closed 2016-07-14 by ruralguru

lutherling12, 2016-07-19:
Which thread is this issue supposed to be in?
How was this issue solved?