mavlink/mavlink #610: Fix Python 3 (Py3K) support

dimasad, 2016-08-14:
According to issue #35, mavlink already supports Python 3. However, installing the master branch from source I could see that many modules are incompatible:
- several incompatible raise statements in `genxmlif` and `minixsv`, generating syntax errors,
- print satements in `genxmlif`, generating syntax errors,
- mavgenerate.py importing `TKinter`, which has been renamed to `tkinter` in Py3K.

Python 2.7 is legacy, now more than ever. We should aim for continuous python 3 support. I can start fixing some of these issues, if more people can help it would be appreciated.

markrjr, 2016-11-06:
Why hasn't this been merged yet?

magicrub, 2016-11-07:
the generator project has moved, it is now at https://github.com/ArduPilot/pymavlink and Python v3 support was added a month ago (and since improved) via PR https://github.com/ArduPilot/pymavlink/pull/10

I suggest this issue be closed.

closed 2017-02-03 by dimasad