mavlink/mavlink #617: common geofence support

dagar, 2016-09-02:
I'd like to add messages to Mavlink common for adding/editing a geofence.

**Option 1**
Move ardupilotmega FENCE_POINT, FENCE_FETCH_POINT, FENCE_STATUS to common.
http://mavlink.org/messages/ardupilotmega#FENCE_POINT
http://mavlink.org/messages/ardupilotmega#FENCE_FETCH_POINT
http://mavlink.org/messages/ardupilotmega#FENCE_STATUS

Questions/issues
 - how do you set the overall altitude
 - delete a point
 - clear all
 - sync reliably

**Option 2**
Create a new set of messages that mirror mission sync, but slightly simplified for a geofence. This would be an opportunity to improve on the above messages if there are any issues.

**Option 2a**
Extend mavlink mission to handle multiple missions (a side benefit), and treat the geofence like a special mission.

FYI @LorenzMeier @DonLakeFlyer @magicrub

DonLakeFlyer, 2016-09-02:
* Overall altitude is GF_MAX_VER_DIST
* You can't delete  a point, you need to resend list again
* In ArduPilot to the total point count is stored in a param. That part is a bit crappy. Clear all sets count param to 0.
* Sync reliably: Currently GCS does a readback.

I forgot about all this stuff with ths ArduPilot messages. I'd go with Option 2.

DonLakeFlyer, 2016-09-02:
Adding @rmackay9 and @tridge

OXINARF, 2016-09-03:
I would go with 2 or 2a. I'm going to do the same (among other things) for the rally point protocol in ArduPilot over the next two months.

DonLakeFlyer, 2016-09-03:
I think I'd rather keep things separate then overload the mission protocol. That said a new protocol which would basically be used to send a set of typed location points to the vehicle could be used for both fence and rally (and cable-cam, ...). And then it really starts to look like mission. Hmm. 

The other question is whether all of the values associated with these things should sent via the new commands as opposed to some parts being in parameters. Having the values in parameters makes it more difficult to save say a fence with full information to a file. You could save parameters to the file as well, but then it doesn't quite have the same send/load from vehicle semantics. So you couldn't say set of a mission where the fence was only in play during the mission. I think it would be valuable for fence and rally. To be able to set them up globally and to include them in a mission.

LorenzMeier, 2016-09-03:
Unless someone can make a clear case why rallye and geofence need a different protocol I feel its time to extend the mission protocol.

DonLakeFlyer, 2016-09-03:
Yeah, I think I managed to talk myself into that as well.

dagar, 2016-09-03:
Works for me. 

I should probably look this up before commenting, but how are components supposed to work? Could these be different components? mission, geofence system, rally, etc

I'd also like the ability to have multiple missions.

LorenzMeier, 2016-09-03:
Should not be different components. We need to extend the mission protocol instead.

WickedShell, 2016-09-27:
If we are extending the mission protocol I'd like to suggest adding a MISSION_REQUEST_CRC and MISSION_CRC that takes an argument of which mission to look at. This would allow us to audit the settings on the vehicle much faster/validate that an upload is correct.

It would also be nice to codify how MISSION_ITEM_INT interacts with the rest of the mission protocol. @meee1 and I made it up when we did the Ardupilot side, and it works well, but it would be worth ensuring that all MAVLink platforms interpret that in the same way.

DonLakeFlyer, 2016-09-27:
FYI: I'll get back to putting together a proposal on this in a few days.

julianoes, 2019-05-08:
As far as I'm aware, this has been added to the mission protocol. I'm closing this.

closed 2019-05-08 by julianoes