mavlink/mavlink #624: Add kinematic uncertainty (e.g., NIC and NACp) to ADSB_VEHICLE message

marclecerf, 2016-09-20:
The ADS-B specification includes requirements for aircraft to transmit a 
quantification of the [uncertainty of their position reports](http://adsb-decode-guide.readthedocs.io/en/latest/nicnac.html).

The current MAVLink message used to pack ADS-B position information received from another aircraft ("ADSB_VEHICLE") does not include the ability to transmit this uncertainty information.

Uncertainty quantification is very important for fusion of ADS-B position reports with other forms of kinematic information, and it should be considered as an improvement to the current ADSB_VEHICLE message to add these fields.

magicrub, 2016-09-20:
We can amend the packet via MAVLink v2.0 but would a flag suffice? If so
that could be added to v1

marclecerf, 2016-09-20:
Hi Tom,

Thanks for the quick response!

Details are below, but in summary, we could likely encode all the uncertainty information in a uint16. (not sure exactly what you meant by a flag? I’m guessing this change could only be incorporated into v2.0.)

I’m also CC’ing Matt and Jeff at uAvionix — I ran into this issue while parsing MAVLink messages from their ADS-B receiver, they might have more insight than me concerning the ADS-B spec. (Matt and Jeff, I found your emails in the uAvionix.xml file.)

The FAA-required uncertainty characterizations are:

- the Navigation Accuracy Category for Position (NACp), 4 bits (values 0-11)
- the Navigation Accuracy Category for Velocity (NACv), 3 bits (values 0-4)
- the Navigation Integrity Category (NIC), 4 bits (values 0-11)
- the Source Integrity Level (SIL), 2 bits (values 0-3)
- the System Design Assurance (SDA), 2 bits (values 0-3)
- (Total = 15 bits)

I’m assuming we wouldn’t need to add any flags saying eg “there is a NACp value for this message” but otherwise, that would be an additional 5 bits, one for each metric.

I’ve been referencing an FAA AC and a public MITRE report:

http://www.faa.gov/documentLibrary/media/Advisory_Circular/AC%2020-165.pdf
https://www.mitre.org/sites/default/files/pdf/10_3026.pdf

Regards,
-Marc

marclecerf, 2016-09-20:
Hey Marc,

Thanks for looping Matt and myself into the conversation.

We've received similar requests from other sources and we're considering
adding a few new MAVLink2 messages to expand the information we're able to
communicate about a vehicle without sending unnecessary data and conserve
the serial bandwidth.

Since this type of extended data isn't necessary for any particular vehicle
until the vehicle is in near proximity or of special interest for some
other reason.  With this is mind, we think adding two new messages to
implement this additional functionality.  One of the new messages would be
a "request" message which the autopilot would send to the ping and should
contain the ICAO for the vehicle in question (plus any other information
that makes sense) and this would elicit the ping to send the other new
message as a response which would contain all of the contents of data
received from the ADSb broadcasts.  This data would contain the NIC,
NAC(s), SIL, SDA and more.

Any comments on this approach?

Jeff W

marclecerf, 2016-09-20:
Hi Jeff,

Yes your suggested implementation sounds like a good way to go.

I agree with you about UQ information (amongst other aircraft details) being important on a case-by-case basis, especially when the aircraft is in close range to the ownship. (Side-note, my application falls in that category, looking to prototype an onboard collision avoidance algorithm.)

So in this case it makes more sense to include the query/response messages as part of the uAvionix XML spec instead of the common XML spec? I’m not familiar with the full vendor list that is using the MAVLink ADS-B message set.

Thanks,
-Marc

LorenzMeier, 2016-09-21:
I think before rushing more content into an uAvionix dialect we should take one step back and quickly touch base with the rest of the industry (e.g. vehicle manufacturers or transponder manufacturers like FLARM).

magicrub, 2016-09-22:
@jwuavionix fyi

hamishwillee, 2020-04-23:
ADSB has moved a long way since this discussion. Closing.

closed 2020-04-23 by hamishwillee