mavlink/mavlink #625: Decalration of MAVLINK_MESSAGE_INFO for dialects

matthewmcgill, 2016-09-21:
When generating the C++ source using the 2.0 generator the MAVLINK_MESSAGE_INFO macro is defined after the base dialects are imported resulting in the MAVLINK_MESSAGE_INFO from common.h being used.  Due to the #ifndef guards it should be defined prior to the includes.

magicrub, 2016-09-21:
Tridge?

LorenzMeier, 2016-09-21:
This is something for @vooon to fix. Can you have a look?

vooon, 2016-09-21:
Please add more info. In *my* C++11 gen that array is placed in dialect namespace and should not cross.

matthewmcgill, 2016-09-21:
Sorry, its in the generated C headers.

I'm looking at the generated ardupilotmega.h. Lines 729-735:

```
// base include
#include "../common/common.h"
#include "../uAvionix/uAvionix.h"

#ifndef MAVLINK_MESSAGE_INFO
#define MAVLINK_MESSAGE_INFO {MAVLINK_MESSAGE_INFO_HEARTBEAT, ...
#endif
```

DonLakeFlyer, 2016-10-05:
I just ran into this as well. This is a significant problem for QGC since it prevents it from being able to show ardupilot messages in the mavlink inspector.

DonLakeFlyer, 2016-10-05:
I also get this when i define MAVLINK_USE_MESSAGE_INFO before mavlink.h include:
```
In file included from ../qgroundcontrol/src/comm/QGCMAVLink.h:20:
In file included from ../qgroundcontrol/libs/mavlink/include/mavlink/v2.0/ardupilotmega/mavlink.h:29:
In file included from ../qgroundcontrol/libs/mavlink/include/mavlink/v2.0/ardupilotmega/ardupilotmega.h:731:
In file included from ../qgroundcontrol/libs/mavlink/include/mavlink/v2.0/ardupilotmega/../common/common.h:1111:
../qgroundcontrol/libs/mavlink/include/mavlink/v2.0/ardupilotmega/../common/../mavlink_get_info.h:11:63: error: unexpected type name 'mavlink_sys_status_t': expected expression
../qgroundcontrol/libs/mavlink/include/mavlink/v2.0/ardupilotmega/../common/common.h:1107:63: note: expanded from macro 'MAVLINK_MESSAGE_INFO'
../qgroundcontrol/libs/mavlink/include/mavlink/v2.0/ardupilotmega/../common/./mavlink_msg_sys_status.h:40:65: note: expanded from macro 'MAVLINK_MESSAGE_INFO_SYS_STATUS'
         { "load", NULL, MAVLINK_TYPE_UINT16_T, 0, 12, offsetof(mavlink_sys_status_t, load) }, \
```

Can't quite figure out what is causing it yet. Wild guess says more header include ordering problems.

DonLakeFlyer, 2016-10-05:
The problem is with "offsetof" usage. If you do this it works:
#define MAVLINK_USE_MESSAGE_INFO
#include <stddef.h>     /* offsetof */
#include "QGCMAVLink.h"

So some sort of include order/missing problem.

tridge, 2016-10-05:
i've reproduced the problem using the test suite and am working on a fix
https://github.com/ArduPilot/pymavlink/issues/13

tridge, 2016-10-05:
i've fixed this in pymavlink master, and fixed the test suite to detect the error

closed 2016-10-05 by tridge

DonLakeFlyer, 2016-10-12:
@tridge Just bumped to latest mav2 headers. Now I'm getting this:
```
In file included from moc/moc_CameraComponent.cpp:9:
In file included from moc/../../qgroundcontrol/src/AutoPilotPlugins/PX4/CameraComponent.h:14:
In file included from ../qgroundcontrol/src/VehicleSetup/VehicleComponent.h:21:
In file included from ../qgroundcontrol/src/Vehicle/Vehicle.h:22:
In file included from ../qgroundcontrol/src/comm/LinkInterface.h:21:
In file included from ../qgroundcontrol/src/comm/QGCMAVLink.h:20:
In file included from ../qgroundcontrol/libs/mavlink/include/mavlink/v2.0/ardupilotmega/mavlink.h:32:
../qgroundcontrol/libs/mavlink/include/mavlink/v2.0/ardupilotmega/ardupilotmega.h:742:10: warning: 'MAVLINK_MESSAGE_INFO' macro redefined [-Wmacro-redefined]
../qgroundcontrol/libs/mavlink/include/mavlink/v2.0/ardupilotmega/../common/common.h:1115:10: note: previous definition is here
1 warning generated.
```