mavlink/mavlink #630: Who controls moving a link to Mav 2.0 protocol? GCS? Vehicle? Both?

DonLakeFlyer, 2016-10-05:
First let me explain the current state of affairs for the two firmware stacks as far as I can tell as well as current QGC implementation:

**ArduPilot - Vehicle side**
* From a parameter you can specify to use mavlink 1 or 2
* If vehicle sees mavlink 2 on link from GCS it will switch to mavlink 2. Not settable. Always on.

**Px4 Pro - Vehicle side**
* From a parameter you can specify to use mavlink 1, 2 or switch to mavlink 2 is GCS sends mavlink 2.

QGroundControl - current implementation
* QGC assumes vehicle has control of mavlink 2 determination. I think this is how MP works as well since I can find any settings in MP to turn on/off mavlink 2 support.
* If QGC sees mavlink 2 coming inbound from vehicle on channel it will switch to mavlink 2 for outbound as well.
* There is no setting in QGC to control mavlink 2 outbound to vehicle.

Problems this causes for ArduPilot:
* Vehicle is set to send mavlink 2
* QGC switches outbound to mavlink 2
* Now try to switch vehicle to mavlink 1
* You can't because QGC is still sending outbound mavlink 2, which forces vehicle back to mavlink 2

So the questions are:
* Is this an ArduPilot only problem?
* Or is the GCS as well supposed to control mavlink 2 sending characteristics? If so, how? Based on an app setting? If so what should those be? Also what about mavlink2 coming back in capabilities. How should ground station respond to that?

DonLakeFlyer, 2016-10-05:
@tridge @LorenzMeier Not sure what is correct here from a GCS standpoint?

tridge, 2016-10-05:
> From a parameter you can specify to use mavlink 1 or 2

note that this is per-UART

> If vehicle sees mavlink 2 on link from GCS it will switch to mavlink 2. Not settable. Always on.

Not true. It won't switch a link to MAVLink2 unless the SERIALn_PROTOCOL setting is 2.

Also note the special behaviour for signing. If there is a valid signing key enabled then it will use MAVLink2 as soon as the link starts, not waiting for a MAVLink2 pkt from GCS.

> Now try to switch vehicle to mavlink 1

what do you mean by this? A SERIALn_PROTOCOL change to 1 won't take effect until you reboot.

> You can't because QGC is still sending outbound mavlink 2, which forces vehicle back to mavlink 2

this will only happen if SERIALn_PROTOCOL=2

DonLakeFlyer, 2016-10-06:
Just FYI: SERIALn_PROTOCOL changes to mavlink 2 seems to take place immediately.

magicrub, 2016-10-07:
Yeah, I thought that was instant too.

LorenzMeier, 2018-03-08:
Current handshaking: https://mavlink.io/en/mavlink_version.html

closed 2018-03-08 by LorenzMeier