mavlink/mavlink #632: No asynchronous framing in mavlink?

mkschreder, 2016-10-07:
I would expect some kind of framing to be present if mavlink is to be used directly on top of a serial link. However including packet start sequences (0xfe) directly into the packet does not result in any escaping of these sequences. This is a serious issue and basically means that messages are missed if mavlink parser gets confused by the extra start sequences. Of course at least the wrong messages are being discarded, but this still means that mavlink itself is never 100% reliable running directly on top of a serial stream with no framing - and therefore should not be advertised as such. 

My test concludes this: 
data: 90 0xfafa
fe 03 00 00 00 64 fa fa 5a e2 c8 fe 03 00 00 00 64 fa fa 5a e2 c8 
rx: 90 0xfafa

With start sequence in data: 
data: 90 0xfefe
fe 03 00 00 00 64 fe fe 5a 6f d9 fe 03 00 00 00 64 fe fe 5a 6f d9 

Feel free to correct me because my tests are just a result of about 1h of fiddling with it. But so far I have not seen any framing and escaping in the packets. 

Suggestion: use PPP style byte stuffed framing always when communicating directly over a serial link.

mkschreder, 2016-10-07:
https://tools.ietf.org/html/rfc1662

LorenzMeier, 2016-10-07:
Since MAVLink has been designed for serial links your base assumption is not correct. What is correct *with the current parser* is that if you receive a half packet you're going to loose the full length of that packet (its going to eat into the next one) plus the remaining bytes of the 2nd packet.

If you have more RAM you can actually keep a ring buffer and try to repeatedly parse the packet to overcome this (its a limitation of the implementation, not the protocol *and* the limitation is intentional to safe buffer space). In practice that's not necessary since what you are describing only happens to the first two packets in a stream of millions.

closed 2016-10-07 by LorenzMeier

mkschreder, 2016-10-07:
Yes but with proper framing you never need to repeatedly parse a packet.
You just need a start byte and an end byte (necessary for triggering a
parse) and to escape these bytes in the data so they never occur in the
data.

What upsets me is that not only the remaining bytes of the packet are lost
when parsing from second byte, but the next packet as well.

mkschreder, 2018-09-10:
> I think this is still an serious issue. Without unique frame delimiter,
> you can theoreticaly lost all next packets forever (in extreme case)
> because of single communication error on unreliable transport medium as
> serial link. It's not just a case it "eat into next one". What if this next
> one contains 0xFE in its payload at the "right" place, so it will eat into
> the next one and so on..? You cannot simply solve this by better parser
> implementation and this is fundamental error of mavlink protocol.
>
> Let's assume this hypothetic mavlink 1 packet (I've made up some random
> IDs and CRC but it's irrelevant here):
> 0xfe 0x02 0x01 0x02 0x03 0x04 0xfe 0x02 0xcc 0xcc
> This packet is repeatedly sent from A to B.
> Now what happen when single error occurs (it can be e.g. single bit error
> in 0xFE start packet marker or lost of some bytes) which will cause B
> packet parser to pick 0xFE from payload part as packet start?
> B will be receiving this invalid packets instead:
> 0xfe 0x02 0xcc 0xcc 0xfe 0x02 0x01 0x02 0x03 0x04
> Indefinitely and will never recover.
>
> In real world situation this probably won't happen. But still situations
> when single error in one packet will result in lost of *many* next
> packets can occur quite easily. Especially when payload contains sequence
> with something like 0xfe 0xff... which can eats into many following valid
> packets because of big false payload length.
>
> Proposed solution with HDLC-like escaping will solve this. There are much
> more interresting byte stuffing schemes like COBS which is nicely
> predictable and could be used as well.

Feel free to use my framing implementation which is based on ppp framing.
It is tested and very lightweight. It will at most ever lose remainder of a
frame and then resync at the beginning of next frame.

Note: to not use framing on a uart connection is just asking for trouble..