mavlink/mavlink #633: MAV_CMD_WAYPOINT_USER_1 is not displayed correctly - #2

Fojtik, 2016-10-09:
Sorry, custom commands still does work not work even in version 3.0.1.
[https://github.com/mavlink/mavlink/issues/525](https://github.com/mavlink/mavlink/issues/525)

magicrub, 2016-10-09:
This command has been add to the protocol correctly 7 months ago so the problem exists in the autopilot implementation. Please close this issue here and make a new issue on the px4/ardupilot issues list for developers, or you, to implement the feature.

Fojtik, 2016-10-09:
Why?
I am not interested in px4/ardupilot implementation. 

I want QGroundControl to display position correctly. Nothing else. QGC displays currently a garbage, all waypoints alligned to a line.

I can provide to anybody IP connection to a simulated plane that uses MAV_CMD_WAYPOINT_USER_1. Very old versions of QGroundcontrol displayed position without any limitation neglectiong to point type.

magicrub, 2016-10-09:
So maybe a problem with qgc? Its not a problem with MAVLink, the protocol had add it. There is nothing to change in this repo to help you.

Fojtik, 2016-10-09:
Sorry, where to post problems with QgroundControl GUI? I guessed that this is right github.
I will repost my issue there and close this.

magicrub, 2016-10-10:
https://github.com/mavlink/qgroundcontrol

Fojtik, 2016-10-11:
Thank you very much, I do not know that there are two different issues list.
I have posted the task here:  [https://github.com/mavlink/qgroundcontrol/issues/4124](https://github.com/mavlink/qgroundcontrol/issues/4124)
And of course I am going to close this.

closed 2016-10-11 by Fojtik