mavlink/mavlink #637: fatal: clone of 'git://github.com/ArduPilot/pymavlink.git' into submodule path 'pymavlink' failed

RickBlacker, 2016-10-13:
Hi all.
I tried to do a clone using this command...

git clone --recursive https://github.com/mavlink/mavlink.git

Everything seemed to work until it got to the ArduPilot submodule.  Has anyone else seen this and been able to work past it?

Thanks!
Rick

splashattack, 2016-10-25:
I too am having this issue. Executing mavgenerate.py will result in the following error:
`Tue Oct 25 test@test-VirtualBox:~/mavlink-master$ python -m mavgenerate
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/test/mavlink-master/mavgenerate.py", line 42, in <module>
    from pymavlink.generator import mavgen
ImportError: No module named pymavlink.generator
`

If anyone has any valuable input it would be much appreciated.

amilcarlucas, 2018-02-19:
This is not a bug. Redo it with master. It works.

IMHO this can be closed now

hamishwillee, 2018-02-19:
Closing, I just retested on master and it is fine. Thanks for testing @amilcarlucas

closed 2018-02-19 by hamishwillee