mavlink/mavlink #639: Current C++ mavlink headers won't compile woth QGroundControl

DonLakeFlyer, 2016-10-18:
Here is my current usage (QGCMAVLink.h):
```
#ifndef QGCMAVLINK_H
#define QGCMAVLINK_H

#define MAVLINK_USE_MESSAGE_INFO
#define MAVLINK_EXTERNAL_RX_STATUS  // Single m_mavlink_status instance is in QGCApplication.cc
#include <stddef.h>                 // Hack workaround for Mav 2.0 header problem with respect to offsetof usage
#include <mavlink_types.h>
extern mavlink_status_t m_mavlink_status[MAVLINK_COMM_NUM_BUFFERS];
#include <mavlink.h>
#endif // QGCMAVLINK_H
```

Which generates the following compile errors:
```
1 warning generated.
In file included from ../qgroundcontrol/src/QGCQuickWidget.cc:11:
In file included from ../qgroundcontrol/src/QGCQuickWidget.h:16:
In file included from ../qgroundcontrol/src/AutoPilotPlugins/AutoPilotPlugin.h:18:
In file included from ../qgroundcontrol/src/VehicleSetup/VehicleComponent.h:21:
In file included from ../qgroundcontrol/src/Vehicle/Vehicle.h:22:
In file included from ../qgroundcontrol/src/comm/LinkInterface.h:21:
In file included from ../qgroundcontrol/src/comm/QGCMAVLink.h:25:
In file included from ../qgroundcontrol/libs/mavlink/include/mavlink/v2.0/ardupilotmega/mavlink.h:32:
../qgroundcontrol/libs/mavlink/include/mavlink/v2.0/ardupilotmega/ardupilotmega.h:745:10: warning: 'MAVLINK_MESSAGE_INFO' macro redefined [-Wmacro-redefined]
../qgroundcontrol/libs/mavlink/include/mavlink/v2.0/ardupilotmega/../common/common.h:1121:10: note: previous definition is here
```

Removing the #include <stdddef.f> workaround for previous offsetof problem not fixed either:
```
In file included from ../qgroundcontrol/src/comm/QGCMAVLink.cc:10:
In file included from ../qgroundcontrol/src/comm/QGCMAVLink.h:24:
In file included from ../qgroundcontrol/libs/mavlink/include/mavlink/v2.0/ardupilotmega/mavlink.h:32:
In file included from ../qgroundcontrol/libs/mavlink/include/mavlink/v2.0/ardupilotmega/ardupilotmega.h:738:
In file included from ../qgroundcontrol/libs/mavlink/include/mavlink/v2.0/ardupilotmega/../common/common.h:1123:
../qgroundcontrol/libs/mavlink/include/mavlink/v2.0/ardupilotmega/../common/../mavlink_get_info.h:10:63: error: unexpected type name 'mavlink_heartbeat_t': expected expression
        static const mavlink_message_info_t mavlink_message_info[] = MAVLINK_MESSAGE_INFO;
                                                                     ^
../qgroundcontrol/libs/mavlink/include/mavlink/v2.0/ardupilotmega/../common/common.h:1121:32: note: expanded from macro 'MAVLINK_MESSAGE_INFO'
../qgroundcontrol/libs/mavlink/include/mavlink/v2.0/ardupilotmega/../common/./mavlink_msg_heartbeat.h:31:69: note: expanded from macro 'MAVLINK_MESSAGE_INFO_HEARTBEAT'
    {  { "custom_mode", NULL, MAVLINK_TYPE_UINT32_T, 0, 0, offsetof(mavlink_heartbeat_t, custom_mode) }, \                                                                    ^
```

icoderaven, 2016-10-19:
I have the same issue with my custom dialect. 

To reconstruct the include error, all I did was change the 
#include <v2.0/common/mavlink.h>
in mavlink_bridge_headers.h to 
#include <v2.0/<custom_dialect_that_overlays_common>/mavlink.h>

icoderaven, 2016-10-19:
@tridge  Is the correct solution for the MAVLINK_THIS_XML_IDX to be set to an autogenerated value, and then MAVLINK_PRIMARY_XML_IDX to be set by the mavlink.h included in the variant?

icoderaven, 2016-10-19:
Aha! I investigated this further (by looking at pymavlink), and I realized that within the update script in the mavlink c_library if we build the definitions for a custom dialect that overlays common, and then also build common, then the generated common files are overwritten, and they are assigned XML_IDXs of 0s, whereas if you omit building the common definitions (which shall be built implicitly), the IDXs are assigned exactly as expected.

closed 2016-10-21 by LorenzMeier

LorenzMeier, 2016-10-21:
For reference: The generator does this:

```
generate_headers ardupilotmega $1
generate_headers autoquad $1
generate_headers matrixpilot $1
generate_headers minimal $1
generate_headers slugs $1
generate_headers test $1
generate_headers ASLUAV $1
```

and this function is this:

```
python pymavlink/tools/mavgen.py \
    --output $CLIBRARY_PATH \
    --lang C \
    --wire-protocol $2.0 \
    message_definitions/v1.0/$1.xml
```

If there is a remaining issue the fix needs to be done in pymavlink.

commit 44133b94 (2016-10-21): Fix message generation by only generating dialects and not common standalone. Fixes #639.