mavlink/mavlink #646: MAVLINK_MESSAGE_INFO redefined

Stifael, 2016-11-08:
With the newest mavlink generator I received the error  "MAVLINK_MESSAGE_INFO" redefined [-Werror] caused by ../mavlink/include/mavlink/v2.0/standard/standard.h.

Stifael, 2016-11-08:
The error occurs when I add a new mavlink message to an existing common.xml file (actually px4). After regeneration, it enters the if statement in standard.h file and redefines all message types. It is strange that the redefinition only happens once a msg is added (erasing the message followed by regeneration still works).

crashmatt, 2017-01-23:
I believe this bug may have been fixed here:
https://github.com/mavlink/mavlink/issues/639

but I am still having the same problem.

crashmatt, 2017-01-23:
Ahh no, don't build common, build standard instead. That builds common but indirectly.

hamishwillee, 2020-04-23:
If this is still actually a problem let me know. Closing.

closed 2020-04-23 by hamishwillee

mnumanuyar, 2021-01-12:
I encountered this issue and fixed it.
I was using a custom dialect and included in ardupilotmega.xml. I was using a sligthly modified version of update_c_library.sh

My problem was : after I generated ardupilotmega.xml, I was generating my xml also. As a result headers related to my xml generated 2 times (once when ı generate for ardupilotmega.xml and once for my xml). and the second time overrides the first time. as a result MAVLINK_THIS_XML_IDX setted to 0 which in-turn caused it to enter mentioned if statement.