mavlink/mavlink #648: How to send receive servo speeds using MavLink?

sytelus, 2016-11-11:
Is there any way to send/receive servo speed (usually in RPM) using  MavLink? I see there is SERVO_OUTPUT_RAW message but that communicates PWM. If my vehicle has sensors to measure RPM, is there any way I can send it out using MavLink message without having to create custom messages?

samacumen, 2016-12-14:
Creating a custom mavlink message is easy if you know exactly how to do it. Add the requirement as enum and/messages in the XML and generate it using Mavgen.py tool. You need to watch out that the libraries are similar on both end nodes (transmit and receive) to avoid incompatibility issues (i.e. the XML files must be generated on both end nodes by the Mavgen tool).

hamishwillee, 2020-04-23:
Closed. Too old.

closed 2020-04-23 by hamishwillee