mavlink/mavlink #649: Mavlink 2 - Dialects

gefink, 2016-11-15:
Hi,

I was wondering what is the proper order to generate the dialects.
A quick example to illustrate the problem.
I have vehicle 1 with dialect 1 that extends common.
I have vehicle 2 with dialect 2 that extends common
I have 1 GS that wants to communicate with both vehicles.
When I generate the dialects I get 3 different common.h (because of the enums).

A work around is to not add any values to enums in the other dialects, but maybe a cleaner solution exists.

Thanks,

Geoff

magicrub, 2016-11-15:
do you mean version instead of dialect? The versions are backwards compatible, and the GCS should detect the version on a per-connection basis.

gefink, 2016-11-16:
No, I'm having a problem with the dialects.  The version detection is working well.

gefink, 2016-11-16:
Maybe I should have said:
vehicle 1 using dialect A that extends common
vehicle 2 using dialect B that extends common

magicrub, 2016-11-16:
you're the first person I've seen who uses the dialects, so I'm not sure. However, there's another way to go about the same thing. Take a look at the uAvionix mfgr file. You could make your own. 
https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/uAvionix.xml

it gets included at the project level like this:
https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/ardupilotmega.xml#L6

gefink, 2016-11-16:
That's interesting.  I didn't think mavlink let you include multiple xmls, but maybe that has changed from v1 to v2?  The uAvionix example doesn't include this problem.
A.xml
`<?xml version="1.0"?>
<mavlink>
<include>common.xml</include>
<enums>
<enum name="MAV_TYPE">
<entry value="50" name="MAV_TYPE_A"/>
</enum>
</enums>
...
</mavlink>
`
B.xml
`<?xml version="1.0"?>
<mavlink>
<include>common.xml</include>
<enums>
<enum name="MAV_TYPE">
<entry value="51" name="MAV_TYPE_B"/>
</enum>
</enums>
...
</mavlink>
`
Then when you generate A.xml, B.xml, and common.xml you will get 3 different common.h
A.xml and B.xml's common.h will not have the enum MAV_TYPE
A.xml's common.h will have the enum MAV_TYPE with 50
B.xml's common.h will have the enum MAV_TYPE with 51
common.xml will have the enum MAV_TYPE with neither 50 nor 51

I realise I can get around the problem by combing A.xml and B.xml into one file, but they are separate.

Another use case for this could be with QGC which currently supports ardupilot and px4.
It would be nice to be able to have
#include<v2.0/standard/mavlink.h>
#include<v2.0/ardupilotmega/mavlink.h>
#include<v2.0/A/mavlink.h>
#include<v2.0/B/mavlink.h>

Hopefully, this explains it better.

Thanks,

Geoff

vooon, 2016-11-16:
You can't use two dialects of C-lib in same program (you will get redefine error). That's why i wrote [C++11 gen][1].

Why not to use one dialect?

[1]: https://github.com/ArduPilot/pymavlink/pull/27

gefink, 2016-11-17:
You are right there probably is no clean way to do this in C.  The c++11 generator is exactly what I'm looking for. Thanks.

gefink, 2016-11-17:
#568

closed 2016-11-17 by gefink