mavlink/mavlink #657: Browse button doesn't work in mavgenerate.py

AlexeyBulatov, 2016-12-01:
Work fine with python3. With python 2.7:

Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1540, in __call__
    return self.func(*args)
  File "./mavgenerate.py", line 130, in browseXMLFile
    xml_file = tkinter.filedialog.askopenfilename(parent=self, title='Choose a definition file')
AttributeError: 'module' object has no attribute 'askopenfilename'

fernandofilhols7, 2017-03-14:
I am having the same problem. Did you have any progress?

Neoster, 2017-03-17:
FYI:
Running: python -i gives me:
Python 2.7.10 (v2.7.10:15c95b7d81dc, May 23 2015, 09:33:12) 
[GCC 4.2.1 (Apple Inc. build 5666) (dot 3)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
MavGenerator works @ my MBPro, browse buttons do too.

magicrub, 2017-03-17:
Hi guys, just wanted to make sure you know know that you're posing an issue about a submodule. If you want it fixed I recommend posting the issue at the source: https://github.com/ArduPilot/pymavlink/issues

amilcarlucas, 2018-02-19:
Please retest with latest master.

And like @magicrub said this is not the correct repository to post this issue to.

IMHO this can be closed.

hamishwillee, 2018-02-20:
I tested this with latest master on Ubuntu Linux 16.04 and Windows 10, on Python 3 (3.5 ubuntu,3.6 Windows) . Worked fine.

Closing.

closed 2018-02-20 by hamishwillee